dsmrreader/dsmr-reader #1517: Minor visuele bug🚨

D1StrX, 2021-11-21:
### Description

[image]

Bij het vergelijken van, in dit geval dagen, geeft dsmr het elektriciteit percentage verschil negatief aan, terwijl dit positief moet zijn. Zie voorbeeld screenshot.
Een minor visuele bug, maar toch net als die de juiste kleur aangeeft. (Icoon is juist)

### DSMR-reader version

4.19

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2021-11-21:
Bedankt voor je melding! Ik zal het meenemen in een volgende release.

dennissiemensma, 2022-02-21:
Wordt opgelost in v5.1. Kwam alleen voor bij het samenvoegen van hoog/laag tarief

closed 2022-02-21 by dennissiemensma

dennissiemensma, 2022-03-01:
Uitgebracht in https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.1.0