dsmrreader/dsmr-reader #1520: 💡Mindergas prognose

rYdQ3J9, 2021-11-22:
### Feature

- Importeren van de mindergas jaarprognose via de API.
- Het zou supermooi zijn als je de mindergas jaarprognose kan terugzien in het DSMR dashboard.

dennissiemensma, 2021-11-22:
Bedankt voor je suggestie! Ik kan me je wens goed voorstellen, alleen beperkt dsmrreader zich tot het lezen/exporteren van de meter en het plotten van "een" eigen visualisatie. Dus eigenlijk de andere kant op, zodat aggregatie van de meter gegevens door andere tools gebruikt kunnen worden voor een andere of aanvullende visualisatie. Vandaar de integratie met bijvoorbeeld minder gas en pvoutput de andere kant op. Als je echt een generiek energy dashboard zoekt zou je naar iets als home assistent kunnen kijken.

rYdQ3J9, 2021-11-23:
Dennis, bedankt voor je uitgebreide antwoord. Ik ga dan verder aan de slag met Home Assistant.

closed 2021-11-23 by dennissiemensma