dsmrreader/dsmr-reader #1521: 🚨column dsmr_mqtt_mqttbrokersettings.qos

Nenodema, 2021-11-24:
### Description

Today, out of the blue dsmr-reader stopped sending mqtt data. When i try to modify the configuration i receive the following errors.

Too many outgoing MQTT messages queued for transit   (now)

### DSMR-reader version

4.17

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: column dsmr_mqtt_mqttbrokersettings.qos does not exist
LINE 1: ...ord", "dsmr_mqtt_mqttbrokersettings"."client_id", "dsmr_mqtt...
                                                             ^
Traceback:
  File "/usr/local/lib/python3.9/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/auth/mixins.py", line 52, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/dsmr/dsmr_frontend/views/configuration.py", line 41, in get_context_data
    mqtt_broker_settings=MQTTBrokerSettings.get_solo(),

  File "/usr/local/lib/python3.9/site-packages/solo/models.py", line 55, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
```

dennissiemensma, 2021-11-24:
Bedankt voor je melding. Kan het zijn dat je een keer een update hebt gedaan die niet helemaal gelukt is/lijkt?

Ik vermoed dat hier een nieuwe versie tegen een oude database aanpraat. Elke nieuwe versie werkt vaak de database zelf bij, al weet ik niet hoe/wanneer dit via de Docker-setup gebeurt.

Als je weet hoe je handmatig in de DSMR-container kan komen, kun je proberen of dit herkend wordt en iets uitvoert:
```
./manage.py migrate
```

Nenodema, 2021-11-24:
En jij bedankt voor je snelle reactie! :-)

Versie 4.17 draait nu al 2 weken perfect, het is vanmorgen, uit het niets fout gegaan. 

manage.py migrate wordt goed uitgevoerd maar er staan geen migraties klaar. Heb je nog een ander idee? 👍

dennissiemensma, 2021-11-24:
Kan het dan nog andersom zijn? Dat je database van DSMR-reader v4.19 is en je container dus van v4.17?

Want dat missende veld is pas in v.419 verwijderd: 9d073581909f8c7e24ecbc9933d5b4f5f61abe6b

Nenodema, 2021-11-24:
Je hebt helemaal gelijk. Ik heb een fout gemaakt met de naam van de images. De oude container stond op "auto start" en de nieuwe container niet. Hierdoor startte de oude container op. *schaam schaam

Thanks!

dennissiemensma, 2021-11-24:
Oke, maar krijg je het nu werkend? 

Ik denk dat je hoe dan ook moet updaten naar v4.19. Daarna kun je eventueel terug naar v4.17 (als dat bewust was). Echter omdat alleen versie 4.19 weet wat er anders is tussen zichzelf en vorige versies (andersom kent v4.17 niet wat in 4.19 zit), kunnen database-downgrades alleen gedaan worden vanaf de versie waar de database al op zit.

dennissiemensma, 2021-11-24:
Voor elke release is er een pin van de DB-migraties: 
- https://github.com/dsmrreader/dsmr-reader/tree/v4/dsmrreader/provisioning/downgrade

Als je terug wilt naar v4.17 moet je iig updaten naar v4.19 en dan dit script (of de commands erin) uitvoeren:
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmrreader/provisioning/downgrade/v4.17.0.sh

Daarna kun je de container weer op v4.17 zetten en zou dat missende veld er weer moeten zijn.

----

Als je gewoon naar v4.19 wil, verwacht ik verder geen issues na de update.

closed 2021-11-24 by dennissiemensma