dsmrreader/dsmr-reader #1523: 🚨 Geen grafiek "Verhouding tarieven" als 100% nachtverbruik.

FredericMa, 2021-11-28:
### Description

Hier in België is gedurende het hele weekend nachttarief van toepassing.
Ik wou mijn verbruik van vorig weekend bekijken en merkte dat de grafiek voor de verhouding van de tarieven niet getoond wordt. Hoewel deze grafiek weinig zou zeggen aangezien deze 100% nachttarief zou aangeven, veronderstel ik wel dat deze zou moeten verschijnen? Ik vermoed dat dit voor 100% dagtarief ook het geval zal zijn maar dat ga ik alvast niet snel te zien krijgen 😃 
[image]

### DSMR-reader version

4.19

dennissiemensma, 2021-11-29:
Bedankt voor je melding, dit is een bug die mij nog niet bekend was. Mocht je bedreven zijn in je browser qua debuggen van netwerkverkeer, dan zie je dat de XHR-call faalt:
```

Server Error
Sorry, something unexpected happened.
Exception:

AttributeError: 'NoneType' object has no attribute 'items'

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 11, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/trends.py", line 111, in get
    end=form.cleaned_data['end_date'],
```

Ik zal kijken of ik dit in een 5.1 of 5.2 release kan fixen.

FredericMa, 2021-11-30:
Geen probleem, jij bedankt voor deze geweldige tool!

FredericMa, 2022-02-16:
Hi Dennis,

Hoewel de visualisatie nu wel een pie toont, is deze wel leeg bij 100% nachtverbruik. Ik had verwacht dat deze een pie weergeeft met 1 element namelijk 100% nachttarief.
Is het de bedoeling dat de chart leeg is?
[image]

dennissiemensma, 2022-02-16:
Bedankt voor je aanvulling. Er zit denk ik nog een bugje in, ik zie deze fout:
```
Uncaught TypeError: data is undefined
    update_trends_tariffs http://rpi4.siemensma.net/static/dsmr_frontend/js/dsmr-reader/trends/tariff.js?r=v5.1:74
    
```

dennissiemensma, 2022-02-18:
Wat verbeteringen klaargezet voor v5.1

closed 2022-02-18 by dennissiemensma

dennissiemensma, 2022-03-01:
Uitgebracht in https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.1.0

FredericMa, 2022-03-01:
Werkt zoals verwacht nu, thanks!