dsmrreader/dsmr-reader #1526: Vrije dataselectie dag/maand/jaar in archief

QuadNL, 2021-11-30:
### Feature

Om makkelijk van een bepaalde data of week een overzicht te krijgen zou het fijn zijn om dit vrij te kunnen selecteren. Nu is er dus enkel keuze uit dag, maand of jaar. 

Geen must, wel een nice to have.

Verder bedankt voor dit mooie project, eenmaal draaiende = geen omkijken naar in werking ervan..

QuadNL, 2021-11-30:
Indien mogelijk zou een vrije dataselectie in het vergelijkingsdeel ook prettig zijn. 
Zo kan je dan eenvoudig twee of meerdere dagen gelijk checken aan elkaar.

dennissiemensma, 2021-12-01:
Bedankt voor je verzoek. Dit is vaker ter sprake gekomen, maar uiteindelijk zit de grootste uitdaging in het vinden van een tool die dergelijke selecties maakt. De wens zelf is namelijk niet heel gek, maar om het ook daadwerkelijk zo te maken dat het voor iedereen genoeg kan en begrijpelijk is is een tweede. https://github.com/dsmrreader/dsmr-reader/issues/1348#issuecomment-923339346

QuadNL, 2021-12-01:
Hi Dennis, 

Bedankt voor je reactie!
Als het technisch te ingrijpend is, vooral niet doen.
DSMRReader werkt nu namelijk prachtig en snel, dat is denk ik ook de kracht in het gebruik ervan. 
Liever simpeler, voldoende informatie en sneller dan het toevoegen van allerhande niet direct nodige zaken en langzamere werking van de applicatie.

closed 2021-12-06 by QuadNL