dsmrreader/dsmr-reader #1527: 🚨 Vergelijk: Geen totaal hoog/laag tarief

QuadNL, 2021-11-30:
### Description

Bij het vergelijken deel is het verbruik en teruglevering enkel hoog en laagtarief zichtbaar, opgeteld totaalverbruik/teruglevering is niet te zien. Wel in archief en dashboard.
[image]

Hier een voorbeeld van uit het archiefgedeelte:
[image]

### DSMR-reader version

4.18

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2021-12-01:
Bedankt voor je feedback. Ik zal er in de toekomst nog een keer wat dieper naar kijken. Puur het optellen zou nog wel toe te voegen zijn.

Overigens kun je voor het vergelijken van achtereenvolgende dagen ook nog de maandweergave in het Archief gebruiken. Dan zie je in 1 oogopslag de verhoudingen tussen alle dagen.

dennissiemensma, 2022-06-17:
In de komende v5.4 release zijn de totalen toegevoegd.

[image]

[image]

closed 2022-06-17 by dennissiemensma