dsmrreader/dsmr-reader #1529: 🙋 Day statistics are lagging behind   (1 day, 16 hours ago)

balk77, 2021-12-04:
### Description

Hi,

I just moved from Home Assistant OS to Home Assistant Supervised, running on Debian 11. I was using the HA addon with success; backup was placed back and all went fine, except.....
`Day statistics are lagging behind   (1 day, 16 hours ago)`

I noticed the following in the log (see below). However, I do see gas readings in the backend. I have also looked at my telegrams; er seems to be a gas reading.

I also had some database corruption (postgresql) on the Home Assistant database which I could resolve by REINDEX'ing. Is there anything I can do on that respect?

Can someone help me? It used to work fine on same hardware.

```
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(211204172858W)
0-0:96.1.1(xxx)
1-0:1.8.1(006224.389*kWh)
1-0:1.8.2(007398.187*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.401*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00000)
0-0:96.7.9(00000)
1-0:99.97.0(1)(0-0:96.7.19)(000101000001W)(2147483647*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(008*A)
1-0:21.7.0(01.664*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxx)
0-1:24.2.1(211204170000W)(05286.174*m3)
!C53B
```
```
2021-12-04 14:02:07,324 ERROR    services     run                              37 | No recent gas reading found
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 69, in export
    last_gas_reading = GasConsumption.objects.filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 325, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 35, in run
    export()
  File "/dsmr/dsmr_mindergas/services.py", line 74, in export
    raise AssertionError(_('No recent gas reading found'))
AssertionError: No recent gas reading found
2021-12-04 15:02:25,164 ERROR    services     run                              37 | No recent gas reading found
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 69, in export
    last_gas_reading = GasConsumption.objects.filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 325, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 35, in run
    export()
  File "/dsmr/dsmr_mindergas/services.py", line 74, in export
    raise AssertionError(_('No recent gas reading found'))
AssertionError: No recent gas reading found
2021-12-04 16:02:41,045 ERROR    services     run                              37 | No recent gas reading found
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 69, in export
    last_gas_reading = GasConsumption.objects.filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 325, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 35, in run
    export()
  File "/dsmr/dsmr_mindergas/services.py", line 74, in export
    raise AssertionError(_('No recent gas reading found'))
AssertionError: No recent gas reading found

2021-12-04 17:02:57,743 ERROR    services     run                              37 | No recent gas reading found
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 69, in export
    last_gas_reading = GasConsumption.objects.filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 325, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 35, in run
    export()
  File "/dsmr/dsmr_mindergas/services.py", line 74, in export
    raise AssertionError(_('No recent gas reading found'))
AssertionError: No recent gas reading found

```

### DSMR-reader version

4.19

### DSMR-reader platform

Other

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                  20.0s / 10.0s / 4032h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             790831
    Consumption records electricity / gas (est.)                                               310826 / 39299

UNRESOLVED ISSUES
    Day statistics are lagging behind                                                     1 day, 18 hours ago
```

dennissiemensma, 2021-12-04:
De foutmelding in de logs over de gasmetingen is vermoedelijk het gevolg van de achterlopende statistieken. Om te zien wat het veroorzaakt kun je het best de debug logging inschakelen. Zie ook andere issues met dezelfde "lagging behind" fouten voor meer informatie over de exacte logging. Als je het niet kan vinden in de debug logs kijk ik wel even mee.

balk77, 2021-12-04:
probleem is dat ik in de addon geen debug aan kan zetten. Ik zal dat eens aan de maintainer vragen. Ik heb even een "gewone" docker DSMR geladen. Helaas geen duidelijke errors in de log, wel rond 20:03 iets. Bedankt voor je hulp!
[_dsmr_logs-1.txt](https://github.com/dsmrreader/dsmr-reader/files/7654512/_dsmr_logs-1.txt)
s

dennissiemensma, 2021-12-04:
Zie mijn eerste comment in #1373 voor wat je zoekt. Ik zie het echter niet zo 123 in je log dump. Je zou in de admin interface bij de scheduled processes kunnen kijken wanneer Generate Day And Hour Statistics gepland staat. Dat is de taak die niet uitgevoerd wordt en zodra dat gebeurt zou die ook moeten loggen waarom.

balk77, 2021-12-05:
resolved! Bedankt. Ik heb een pagina aangemaakt in de docs. Zie #1530

closed 2021-12-05 by balk77

dennissiemensma, 2021-12-07:
Bedankt voor je aanvulling! Ik ben even niet in de gelegenheid om het in detail te bekijken maar ik kom er zeker op terug!