dsmrreader/dsmr-reader #1531: 🙋MQTT Broker sends wrong dsmr/json data message

JanCLinux, 2021-12-07:
### Description

Hello,
I try to exchange data from DSMR to Influxdb by using the MQTT Broker
The MQTT Broker indeed sends the telegram data but only three data points are received by Influxdb.
I checked the messages from the MQTT Broker with MQTTLens and it appeared that three data points at the end  that are correctly received are sent without 'quotes' while all other values are contained in 'quotes'.

Message captured bij Google MQTTLens:
  "id": 164052,
  "timestamp": "2021-12-07T17:02:08Z",
  "electricity_delivered_1": "5834.163",
  "electricity_returned_1": "424.997",
  "electricity_delivered_2": "4637.053",
  "electricity_returned_2": "988.696",
  "phase_currently_delivered_l1": "0.726",
  "phase_currently_delivered_l2": "0.252",
  "phase_currently_delivered_l3": "0.117",
  "phase_currently_returned_l1": "0.000",
  "phase_currently_returned_l2": "0.000",
  "phase_currently_returned_l3": "0.000",
  "phase_voltage_l1": "227.3",
  "phase_voltage_l2": "224.9",
  "phase_voltage_l3": "227.7",
  "phase_power_current_l1": 3,
  "phase_power_current_l2": 1,
  "phase_power_current_l3": 0

How can this be resolved?

Many thanks in advance,
Jan

### DSMR-reader version

4.19

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2021-12-07:
Hi this is probably caused by the Django serializer. See #1319 for more details. You can either convert the strings somewhere outside dsmrreader or you can try overriding the setting mentioned in one of the comments of #1319. This affects the API response as well.https://github.com/dsmrreader/dsmr-reader/issues/1319#issuecomment-815276838

JanCLinux, 2021-12-07:
Thanks, I gave it a try to amend the settings.py file according #1319, but unfortunately the problem persists.
Furthermore to me it's quite strange that 3 of the data points and the id are actually presented as floats/numbers while the others as strings.
There seems to be some inconsistency in the coding.

dennissiemensma, 2021-12-07:
The 3 numbers are integers and therefore not causing floating point issues. Decimals however require specific handling to not lose precision. Encoding them as strings is quite a common strategy, although it may seem weird when you've never dealt with the side effects.  That being said, I do not have another easy fix for you. I did a quick check and dsmrreader uses the default Django serializer, https://github.com/dsmrreader/dsmr-reader/blob/c037848e0f96028fb500415b9289df40f81bc14f/dsmr_mqtt/services/callbacks.py#L187, so I'm not sure how to fix it properly for you on short term, if at all.

JanCLinux, 2021-12-07:
Thanks for your explanation Dennis. 
I will try to find a different way to exchange data.

Best regards, Jan

dennissiemensma, 2021-12-08:
I almost forgot, you can try the (shallow) dsmrreading exporter to influxdb as well. It's not heavily documented, but you should be able to find it in the admin settings. Only supports influx 1.x though. 2.x will come next major dsmrreader bump.

closed 2021-12-16 by dennissiemensma