dsmrreader/dsmr-reader #1532: 🤔Verkeerder versie ?

Marchellos, 2021-12-08:
### Your question

Na een paar dagen wamen er geen gegevens meer binnen.
Als ik terug lees dan lijkt het probleem zich eerder te hebben voorgedaan.
Maar ik had de indruk dat het probleem was opgelost.
Of heb ik misschien de verkeerde versie geïnstalleerd ?
Welke versie moet ik installeren voor Home Assistant (V5) ?

dennissiemensma, 2021-12-08:
Bedankt voor je melding. Ik heb echt veel meer informatie en context nodig want ik weet eerlijk gezegd niet wat je bedoelt. Dsmrreader is momenteel nog versie 4. Heb je een debug info dump? Heb je debug logging aangezet en de logfiles bekeken?

Marchellos, 2021-12-09:
"Na een paar dagen kwamen er geen gegevens meer binnen".
Hiermee bedoelde ik dat de verbruik cijfers (kW) na 2-3 dagen niet meer werden bijgewerkt.

Ik heb nu de DSMR reader een keer verwijderd en opnieuw geïnstalleerd.
Ik ben nog veel dingen binnen Home Assistant aan het uitzoeken/uitproberen.
Ik zal het komende dagen opvolgen of de situatie zich weer voordoet.
Zo ja dan zal ik een uitgebreidere rapportage geven. 
Zo niet dan zal ik  zondag avond deze rapportage sluiten.

Marchellos, 2021-12-10:
Deze meldingen zag ik vanmorgen in het logboek:

Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.9/site-packages/dsmr_parser/clients/protocol.py:117
First occurred: 9 december 2021 18:12:30 (321 occurrences)
Last logged: 09:15:06

keep-alive check failed
----------------------------------------------------------------------------------------------------------------
Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.9/site-packages/dsmr_parser/clients/protocol.py:136
First occurred: 9 december 2021 19:48:45 (3 occurrences)
Last logged: 9 december 2021 19:56:11

Invalid telegram. The CRC checksum '14567' does not match the expected '51379'
Invalid telegram. The CRC checksum '44979' does not match the expected '52540'
Invalid telegram. The CRC checksum '18924' does not match the expected '62297'

dennissiemensma, 2021-12-10:
Kan het zijn dat je geen dsmrreader gebruikt maar dsmr_parser? Ik zie namelijk die referentie in de melding. De fout zelf gaat overigens over CRC checksums. Zolang het niet constant voorkomt zal dat niet de oorzaak zijn.

dennissiemensma, 2021-12-15:
@Marchellos ben je nog wat wijzer geworden?

Marchellos, 2021-12-15:
Hallo Dennis,

Het betreft [DSMR integratie voor  Home Assistan](https://www.home-assistant.io/integrations/dsmr/)t  🤭
Zojuist logbestand nog eens gecontroleerd, maar geen enkele fout melding meer binnen gekregen.
De indruk bestaat dat het met instellingen op de Proxmox server te maken heeft gehad.
Nadat er settings op de Pproxmox waren gewijzigd heb ik geen foutmeldingen meer gekregen.
Welke instellingen dat waren weet ik niet want de Proxmox heb ik zelf niet in beheer.
Ik ga er dus vanuit dat het probleem is opgelost.
Dank voor je bericht en belangstelling.

closed 2021-12-15 by Marchellos