dsmrreader/dsmr-reader #1533: 💡phase_power_current = integer

mrombouts, 2021-12-08:
### Feature

Hallo, 

Is er een specifieke reden waarom phase_power_current door dsmr Reader als iteger wordt behandelt? 
Het telegram geeft dit nochtans als een floating point waarde:

/FLU5\253769484_A

1-0:31.7.0(001.06*A)
1-0:51.7.0(002.07*A)
1-0:71.7.0(001.47*A)

Hierdoor wordt deze waarde in mqtt en verdere toepassingen (Home Assistant) ook als integer gezien. 

Is het mogelijk dit te veranderen naar floating point? 

Bedankt, 
Michiel

dennissiemensma, 2021-12-08:
Bedankt voor je melding. Je bent de eerste waarbij ik dit zie of het iemand opvalt. Voor de NL varianten zijn het sowieso integers. De testdata voor Fluvius (die ik ooit van iemand gehad heb) gebruikt ook integers: https://github.com/dsmrreader/dsmr-reader/blob/c037848e0f96028fb500415b9289df40f81bc14f/dsmr_datalogger/tests/datalogger/test_fluvius_polyphase.py#L41. Dus ik ben benieuwd of dit een nieuwe variant is. In principe kan ik het datatype wel een keer aanpassen, maar dat is zeker niet triviaal omdat het alle andere gebruikers dan ook zou raken.

mrombouts, 2021-12-08:
Hallo Dennis, 

Dit hoeft toch niet iedereen te raken? Als je bv kijkt naar het inkomende telegram, en dan afhankelijk van het formaat interger of FP gebruikt? 

Alvast bedankt voor je snelle reactie!

dennissiemensma, 2021-12-08:
Het raakt daarmee de API, mqtt, influxdb allemaal omdat de serializers kijken naar het type en niet de waarde. En dus exporteren er dan andere types. Binnen dsmrreader zelf raakt het inderdaad relatief weinig.

mrombouts, 2021-12-09:
ok, dankjewel voor de reactie!

eTh0maz, 2022-03-02:
Ik stel hetzelfde vast, de waarden na het decimaal punt gaan verloren.
De Fluvius meter is hier ingebruik genomen voorjaar 2020.
[image]

dennissiemensma, 2022-03-03:
Het valt me nu pas op dat in dit issue hetzelfde speelt als in #1394.

Ik sluit de huidige, want dan pas ik de andere wel even aan.

dennissiemensma, 2022-03-03:
# Zie #1394

closed 2022-03-03 by dennissiemensma