dsmrreader/dsmr-reader #1534: 🚨Periode energiecontracten mist laatste dag

thijskuijpers, 2021-12-08:
### Description

De berekening van totalen over de contractperiode lijkt de laatste dag te missen. De einddatum kan niet als begindatum van de volgende periode worden gebruikt, dus ik verwacht "van <datum> t/m <datum>" i.p.v. "van <datum> tot <datum>".
Zie screenshot, waarin ik 2 dagen verwacht maar alleen de eerste dag wordt meegenomen in de berekening. 
[image]

### DSMR-reader version

4.19

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2021-12-08:
Bedankt voor je melding. Ik zal hier later wat dieper naar kijken. Deze bug was mij nog niet bekend.

hv92, 2022-01-12:
Hier hetzelfde probleem. Duidelijk te zien dat maandag 3 januari (hoogtarief) niet mee wordt gerekend.

[image]

dennissiemensma, 2022-01-13:
Ik heb de oorzaak in ieder geval al gevonden:

https://github.com/dsmrreader/dsmr-reader/blob/c037848e0f96028fb500415b9289df40f81bc14f/dsmr_stats/services.py#L270

- `day__gte=start` = Groter dan of gelijk aan startdatum
- `day__lt=end` = Kleiner dan einddatum

Subtiel en is nooit opgevallen, het moet dus `day__lte=end` zijn zodat de einddatum ook meetelt. Ik neem het mee in een release.

dennissiemensma, 2022-01-18:
Oh ik bedoelde ook niet dat die regel aangepast moest worden, maar het herleidt wel naar hoe het komt. De uiteindelijke fix is om de plek die dat aanroept aan te passen zodat die altijd +1 dag doet. Want bovenstaande query wordt inderdaad ook door andere features gebruikt.

dennissiemensma, 2022-01-18:
Dit is de plek waar het vermoedelijk gefixt mag worden: https://github.com/dsmrreader/dsmr-reader/blob/c037848e0f96028fb500415b9289df40f81bc14f/dsmr_consumption/services.py#L514

Naar zoiets:
```
        end_date = current.end or timezone.now().date()
        summary = dsmr_stats.services.range_statistics(
            start=current.start,
            end=end_date + timezone.timedelta(days=1)
        )
```

Ik kan het nu niet makkelijk testen, omdat ik qua ontwikkelomgeving nog midden in de voorbereidingen voor v5 zit.

hv92, 2022-01-18:
Thanks! Ik had nav je vorig bericht ook al die regel gevonden en aangepast. Ik zal terugkoppelen of die wijziging de fix is, er zit volgens mij nog ergens een cronjob oid die periodiek de data update ;)

dennissiemensma, 2022-01-20:
Ik neem het mee in v4.20. Fix via 53dfa85f1a129efe24234fc29bcdf45884bd7393.

closed 2022-01-20 by dennissiemensma