dsmrreader/dsmr-reader #1535: 🙋Temperatuur grafiek wordt niet gevuld; reboot/restart container lost het "soms" op

JonKlaus, 2021-12-14:
### Description

De temperatuur grafiek op basis van buienradar wordt af en toe niet meer gevuld. 
[image]
Het is een probleem dat af en toe optreedt, meestal nadat een DSMR update doorgevoerd heb en de dsmr-reader en postgress containers herstart heb. Het rare: soms lost een reboot van de onderliggende raspberry pi of een restart van de containers het op, waarna de grafiek netjes gevuld wordt. Dit werkt echter niet altijd.
Als het WEL werkt, dan wordt de grafiek vervolgens feilloos gevuld tot de volgende restart (lees: app update). Daarna is het weer een gok of het werkt.
Info geeft een "Alles lijkt soepel te draaien." terug, als ik de logs uit de container trek zie ik 1-2-3 ook niets geks. Hoe verder; debug logging inschakelen?

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                             252088
    Consumption records electricity / gas (est.)                                                 50814 / 5707
```

dennissiemensma, 2021-12-14:
Bedankt voor je vraag. Ik zie geen fouten in je dump dus wellicht dat de logfiles wat zeggen. Als daar niets in staat dan kun je debug logging inschakelen https://github.com/xirixiz/dsmr-reader-docker#dsmr-related-defaults-are-shown-as-value. DSMRREADER_LOGLEVEL=DEBUG

JonKlaus, 2021-12-15:
Ik zie maar 3 CRIT regels; DEBUG staat aan, de eerste 7 minuten heb ik bijgesloten
[debuglog.txt](https://github.com/dsmrreader/dsmr-reader/files/7718346/debuglog.txt)
.

JonKlaus, 2021-12-15:
Hop, daar is ie:

```
2021-12-15 12:05:02,978 ERROR    services     run                              22 | Buienradar: Failed to read API: HTTPSConnectionPool(host='data.buienradar.nl', port=443): Max retries exceeded with url: /2.0/feed/json (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f8d0e86d0>: Failed to establish a new connection: [Errno -3] Try again'))
2021-12-15 12:05:02,983 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Weather update" to 2021-12-15 12:05:02.978556+00:00 (ETA 0:59:59.995504)

```

dennissiemensma, 2021-12-15:
Bedankt voor je aanvulling. De fout lijkt gerelateerd aan ofwel de Buienradar API ofwel de verbinding er naar toe.

Het is mij wel bekend dat de API niet altijd reageert, maar dat zou incidenteel moeten zijn. Nu lijkt die het namelijk gewoon te doen: https://data.buienradar.nl/2.0/feed/json

Ik heb hier helaas niet echt een oplossing voor of het moet in je eigen netwerk of Docker zitten. Je zou nog kunnen kijken naar iets van firewalls ofzo? Helemaal omdat je er structureel last van lijkt te hebben en niet indicenteel. Want in mn eigen installatie mis ik geen temperatuurmetingen in de afgelopen twee dagen (wat niet alles zegt, maar ook niet structureel mis gaat bij mij dus).

En anders zou je nog Xirixiz kunnen vragen, maar ik weet niet hoe diep hij in dit soort issues zit. Je kunt ook nog even googlen naar andere oorzaken oid: https://www.google.com/search?q="Failed+to+establish+a+new+connection"+try+again+docker

JonKlaus, 2021-12-16:
Hoi Dennis, dank voor de terugkoppeling. Als ik wat dieper de debug logging in duik zie ik dat de dropbox backup connectie met een soortgelijke foutmelding terugkomt. 
```2021-12-16 09:58:22,352 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Dropbox export" (dsmr_dropbox.services.run)
2021-12-16 09:58:27,370 ERROR    schedule     execute_scheduled_processes      37 | (ConnectionError) dsmr_dropbox.services.run errored: HTTPSConnectionPool(host='api.dropboxapi.com', port=443): Max retries exceeded with url: /2/users/get_space_usage (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f8cc0f250>: Failed to establish a new connection: [Errno -3] Try again'))
2021-12-16 09:58:27,376 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Dropbox export" to 2021-12-16 08:58:57.371197+00:00 (ETA 0:00:29.994549)
```
Lijkt dus inderdaad een (uitgaand) connectivity probleem, niet een buienradar probleem. Vanaf andere hosts op mijn netwerk kom ik ook prima bij de buienradar API, dus geen generiek netwerk probleem. Ik ga eens wat dieper de docker config induiken en zal laten weten of ik wat vind. Thanks voor de tips tot dusverre!