dsmrreader/dsmr-reader #1541: 🙋Elke nacht crasht mijn dsmr reader maar geen idea waarom.

Diondk, 2021-12-21:
### Description

Elke ochtend als ik inlog op mijn server is mijn dsmr reader container gestopt. Dit is eigenlijk al zo sinds de laatste update, gisteren heb ik mijn externe postgres geupgrade naar versie 14 omdat ik meerdere containers heb die daar van gebruik maken.

Nu heb ik de volgende errors:

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

could not find a "pg_dump" to execute
pg_dump: error: server version: 14.1 (Debian 14.1-1.pgdg110+1); pg_dump version: 13.5

pg_dump: error: aborting because of server version mismatch

2021-12-21 08:53:12,788 ERROR schedule execute_scheduled_processes 37 | (AttributeError) dsmr_backup.services.backup.run errored: 'NoneType' object has no attribute 'read'

2021-12-21 08:53:13,789 INFO success: processes entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2021-12-21 08:53:13,790 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2021-12-21 08:53:15,792 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2021-12-21 08:53:17,793 INFO success: nginx entered RUNNING state, process has stayed up for > than 5 seconds (startsecs)
could not find a "pg_dump" to execute
pg_dump: error: server version: 14.1 (Debian 14.1-1.pgdg110+1); pg_dump version: 13.5

pg_dump: error: aborting because of server version mismatch

```
2021-12-21 08:53:43,795 ERROR schedule execute_scheduled_processes 37 | (AttributeError) dsmr_backup.services.backup.run errored: 'NoneType' object has no attribute 'read'

could not find a "pg_dump" to execute
pg_dump: error: server version: 14.1 (Debian 14.1-1.pgdg110+1); pg_dump version: 13.5

pg_dump: error: aborting because of server version mismatch

2021-12-21 08:54:13,811 ERROR schedule execute_scheduled_processes 37 | (AttributeError) dsmr_backup.services.backup.run errored: 'NoneType' object has no attribute 'read'
```
Moet ik nu terug naar een oudere versie van postgres om deze error op te lossen?

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             426229
    Consumption records electricity / gas (est.)                                                32361 / 26594
```

Diondk, 2021-12-21:
De error is weg nadat ik de backup heb uitgeschakeld, aangezien mijn postgres extern draait en daar een backup van wordt gemaakt. Ben ik dus wel veilig. Dan kan deze dicht, en zal ik morgen is kijken of hij nog crasht. mocht dat zijn dan maak ik een nieuw issue aan.

closed 2021-12-21 by Diondk

dennissiemensma, 2021-12-21:
@Diondk bedankt voor je melding. Ik denk dat dit komt doordat de DB-client (`pg_dump`) een oudere versie is.

```
pg_dump version: 13.5
server version: 14.1
```

Je kunt het ook oplossen door de DB-client te upgraden naar versie 14, alleen is dat wellicht minder triviaal via een Docker image.