dsmrreader/dsmr-reader #1542: 🚨Datalogger: Netwerk socket verbreekt continue verbinding

ualex73, 2021-12-21:
### Description

Ik heb nu de DSMR Reader gekoppeld via een netwerk socket, via de slimmelezer++ van Marcel. Ik zie in de GUI en logging geen problemen, maar ik merk op dat de datalogger continue de netwerk verbinding verbreekt (hierdoor verliezen we dus telegrammen)

Bijgesloten de tcpdump via met Wireshark te openen is:
[p23.zip](https://github.com/dsmrreader/dsmr-reader/files/7756550/p23.zip)
.

DSMR Reader IP: 172.19.0.3
ESP-Link: 192.168.12.3

tcp.stream eq 0:
packet 144: DSMR Reader stuurt een FIN naar de ESP-Link om de verbinding te verbreken (dus DSMR verbreekt het, niet de ESP-Link!).

### DSMR-reader version

4.19.0

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                                365
    Consumption records electricity / gas (est.)                                                      -1 / -1

UNRESOLVED ISSUES
    Geen dagstatistieken gevonden                                                                          nu
```

dennissiemensma, 2021-12-21:
Bedankt voor je melding. DSMR-reader werkt niet (qua code) op pakket-niveau, dus vermoedelijk is er iets anders aan de hand wat dit veroorzaakt. 

Je kunt het beste beginnen met het inschakelen van DEBUG-logging en daarna kijken in de logfile van de datalogger. Je kunt dan exact zien wat die doet, wat die leest, etc. Mogelijk staat daar al een fout of oorzaak in.

ualex73, 2021-12-21:
Hoi Dennis, ik had al de datalogger op debug gezet. De logfile is als volgt:
[dsmr-datalogger.txt](https://github.com/dsmrreader/dsmr-reader/files/7758006/dsmr-datalogger.txt)

Ik zie dat de datalogger het volgende lijkt te doen:
1. connect naar het IP/port
2. verwerkt een telegram (een volledige, indien de eerste fout is - pakt die de 2e)
3. Gaat 5 seconden slapen
4. Reconnect weer naar het IP/port - dus de verbinding wordt niet hergebruikt, maar een nieuwe opgebouwd

dennissiemensma, 2021-12-21:
Ah, ik heb het al gevonden, blijkbaar heb ik dit ooit ingesteld: https://github.com/dsmrreader/dsmr-reader/blob/c037848e0f96028fb500415b9289df40f81bc14f/dsmr_datalogger/management/commands/dsmr_datalogger.py#L31-L33

Je kunt proberen de sleep te verlagen.

dennissiemensma, 2021-12-21:
#1114

dennissiemensma, 2021-12-21:
Het was even zoeken, maar nu staat die op 1: https://github.com/dsmrreader/dsmr-reader/commit/fc5b9ab0f5924a35a68c61af632f80938854c1a2#diff-bb441a5dc32bf8d5bf5eae25e088b8726015d36d316824dcce055b0f6dfa1b44R115

Dus probeer een datalogger-sleep van 1 of korter.

ualex73, 2021-12-21:
Thanks, ik ga 0 en 1 morgen proberen!

dennissiemensma, 2021-12-21:
Het is trouwens ook een optie om ze gewoon op 5 of zelfs 10 te zetten. Je miste door de sleep sowieso telegrammen, maar voor het bijhouden van je verbruik kun je prima wat telegrammen missen hoor. 
Ik zou de minimale sleep nog (wat) hoger kunnen zetten, maar voor software is een seconde op een lokaal netwerk al een eeuwigheid, vandaar de reconnect.

Als je niets wilt missen, zul je sowieso `0` moeten instellen denk ik. Alleen houd er wel rekening mee dat de datagrootte en op termijn performance daaronder lijdt, tenzij je retentie/opschoning goed hebt ingesteld. Voor de nauwkeurigheid in DSMR-reader zelf maakt het echter amper uit.