dsmrreader/dsmr-reader #1543: 🙋Elke nacht om 3 uur kapt dsmr er mee, geen idee waarom

Diondk, 2021-12-22:
### Description

Elke nacht om 3 uur kapt dsmr er mee, dan loopt wel de backup van de containers maar tot de vorige update was dit nooit een probleem. nu dus opeens wel, geen idee waarom dit zie ik in mijn container logs staan.

````[2021-12-21 13:21:18 +0100] [84] [INFO] Booting worker with pid: 84
[2021-12-21 13:29:36 +0100] [84] [INFO] Autorestarting worker after current request.
[2021-12-21 13:29:36 +0100] [84] [INFO] Worker exiting (pid: 84)
[2021-12-21 13:29:37 +0100] [85] [INFO] Booting worker with pid: 85
[2021-12-21 13:37:54 +0100] [85] [INFO] Autorestarting worker after current request.
[2021-12-21 13:37:54 +0100] [85] [INFO] Worker exiting (pid: 85)
[2021-12-21 13:37:55 +0100] [86] [INFO] Booting worker with pid: 86
[2021-12-21 13:46:12 +0100] [86] [INFO] Autorestarting worker after current request.
[2021-12-21 13:46:12 +0100] [86] [INFO] Worker exiting (pid: 86)
[2021-12-21 13:46:13 +0100] [87] [INFO] Booting worker with pid: 87
[2021-12-21 13:54:31 +0100] [87] [INFO] Autorestarting worker after current request.
[2021-12-21 13:54:31 +0100] [87] [INFO] Worker exiting (pid: 87)
[2021-12-21 13:54:32 +0100] [88] [INFO] Booting worker with pid: 88
[2021-12-21 14:02:49 +0100] [88] [INFO] Autorestarting worker after current request.
[2021-12-21 14:02:49 +0100] [88] [INFO] Worker exiting (pid: 88)
[2021-12-21 14:02:50 +0100] [89] [INFO] Booting worker with pid: 89
[2021-12-21 14:11:08 +0100] [89] [INFO] Autorestarting worker after current request.
[2021-12-21 14:11:08 +0100] [89] [INFO] Worker exiting (pid: 89)
[2021-12-21 14:11:09 +0100] [90] [INFO] Booting worker with pid: 90
[2021-12-21 14:19:26 +0100] [90] [INFO] Autorestarting worker after current request.
[2021-12-21 14:19:26 +0100] [90] [INFO] Worker exiting (pid: 90)
[2021-12-21 14:19:27 +0100] [91] [INFO] Booting worker with pid: 91
[2021-12-21 14:27:44 +0100] [91] [INFO] Autorestarting worker after current request.
[2021-12-21 14:27:44 +0100] [91] [INFO] Worker exiting (pid: 91)
[2021-12-21 14:27:45 +0100] [92] [INFO] Booting worker with pid: 92
[2021-12-21 14:36:03 +0100] [92] [INFO] Autorestarting worker after current request.
[2021-12-21 14:36:03 +0100] [92] [INFO] Worker exiting (pid: 92)
[2021-12-21 14:36:03 +0100] [93] [INFO] Booting worker with pid: 93
[2021-12-21 14:44:20 +0100] [93] [INFO] Autorestarting worker after current request.
[2021-12-21 14:44:20 +0100] [93] [INFO] Worker exiting (pid: 93)
[2021-12-21 14:44:20 +0100] [94] [INFO] Booting worker with pid: 94
[2021-12-21 14:52:39 +0100] [94] [INFO] Autorestarting worker after current request.
[2021-12-21 14:52:39 +0100] [94] [INFO] Worker exiting (pid: 94)
[2021-12-21 14:52:40 +0100] [95] [INFO] Booting worker with pid: 95
[2021-12-21 15:00:56 +0100] [95] [INFO] Autorestarting worker after current request.
[2021-12-21 15:00:56 +0100] [95] [INFO] Worker exiting (pid: 95)
[2021-12-21 15:00:57 +0100] [96] [INFO] Booting worker with pid: 96
[2021-12-21 15:09:14 +0100] [96] [INFO] Autorestarting worker after current request.
[2021-12-21 15:09:14 +0100] [96] [INFO] Worker exiting (pid: 96)
[2021-12-21 15:09:14 +0100] [97] [INFO] Booting worker with pid: 97
[2021-12-21 15:17:33 +0100] [97] [INFO] Autorestarting worker after current request.
[2021-12-21 15:17:33 +0100] [97] [INFO] Worker exiting (pid: 97)
[2021-12-21 15:17:33 +0100] [98] [INFO] Booting worker with pid: 98
[2021-12-21 15:25:51 +0100] [98] [INFO] Autorestarting worker after current request.
[2021-12-21 15:25:51 +0100] [98] [INFO] Worker exiting (pid: 98)
[2021-12-21 15:25:51 +0100] [99] [INFO] Booting worker with pid: 99
[2021-12-21 15:34:09 +0100] [99] [INFO] Autorestarting worker after current request.
[2021-12-21 15:34:09 +0100] [99] [INFO] Worker exiting (pid: 99)
[2021-12-21 15:34:10 +0100] [100] [INFO] Booting worker with pid: 100
[2021-12-21 15:42:27 +0100] [100] [INFO] Autorestarting worker after current request.
[2021-12-21 15:42:27 +0100] [100] [INFO] Worker exiting (pid: 100)
[2021-12-21 15:42:27 +0100] [101] [INFO] Booting worker with pid: 101
[2021-12-21 15:50:45 +0100] [101] [INFO] Autorestarting worker after current request.
[2021-12-21 15:50:45 +0100] [101] [INFO] Worker exiting (pid: 101)
[2021-12-21 15:50:45 +0100] [102] [INFO] Booting worker with pid: 102
[2021-12-21 15:59:03 +0100] [102] [INFO] Autorestarting worker after current request.
[2021-12-21 15:59:03 +0100] [102] [INFO] Worker exiting (pid: 102)
[2021-12-21 15:59:03 +0100] [103] [INFO] Booting worker with pid: 103
[2021-12-21 16:07:21 +0100] [103] [INFO] Autorestarting worker after current request.
[2021-12-21 16:07:21 +0100] [103] [INFO] Worker exiting (pid: 103)
[2021-12-21 16:07:21 +0100] [104] [INFO] Booting worker with pid: 104
[2021-12-21 16:15:39 +0100] [104] [INFO] Autorestarting worker after current request.
[2021-12-21 16:15:39 +0100] [104] [INFO] Worker exiting (pid: 104)
[2021-12-21 16:15:39 +0100] [105] [INFO] Booting worker with pid: 105
[2021-12-21 16:23:57 +0100] [105] [INFO] Autorestarting worker after current request.
[2021-12-21 16:23:57 +0100] [105] [INFO] Worker exiting (pid: 105)
[... 282 lines cut ...]
2021-12-22 03:05:02,649 INFO stopped: processes (terminated by SIGTERM)```

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             418524
    Consumption records electricity / gas (est.)                                                31594 / 26594

UNRESOLVED ISSUES
    Geen recente metingen ontvangen                                                             4 uur geleden
```

dennissiemensma, 2021-12-22:
Het kan komen door #1541. Het lijkt erop dat de DB afsluit: "terminating connection due to administrator commandserver closed the connection unexpectedly
This probably means the server terminated abnormally
before or while processing the request."

Diondk, 2021-12-22:
Ik had dit ook al voor dat ik mijn database heb geupgrade naar versie 14, maar zal mijn backup vanavond is uitzetten op mijn containers misschien dat hij het morgen nog doet dan.

dennissiemensma, 2021-12-22:
Je kunt dit dan het beste bij xirixiz voorleggen. Ik weet namelijk niet of het dsm gerelateerd of doxker gerelateerd is eerlijk gezegd.

Diondk, 2021-12-25:
Na het uitzetten van de backup voor de database container crasht hij inderdaad niet meer, raar dat het hiervoor nooit gebeurde. Zal na de kerst een issue maken xirixiz.

dennissiemensma, 2021-12-25:
Dank voor je snelle update! 

De oorzaak is de Postgres-versie-mismatch, maar het gevolg (dat alles stopt) is wellicht een bijeffect van hoe processen nu intern draaien in de Docker-versie. https://github.com/xirixiz/dsmr-reader-docker/issues/241#issuecomment-999032323

closed 2021-12-25 by dennissiemensma