dsmrreader/dsmr-reader #1545: 🤔 MQTT berichten dubbel?

raymonvdm, 2021-12-30:
### Your question

Vandaag ben ik weer in DSMR Reader gedoken nadat de P1 kabel is vervangen door een P1-Wifi module (omdat de pc waar dez e aanhing door ging) maar nu heb ik denk ik een bug geraakt met MQTT

Dit is wat ik zie op het moment dat ik kijk met 
```
mosquitto_sub -h mqttserver -t "dsmr/" -u "user" -P "password"
```

```
45733327
1.289
1.289
6423.138
2021-12-30T17:22:02Z
5.069
5.069
5.058
5.058
5.047
5.047
5.036
5.036
2021-12-11T06:50:00+01:00
5.025
5.025
45733326
1.305
1.305
6423.136
2021-12-30T17:21:56Z
5.013
5.013
5.002
5.002
4.991
4.991
4.980
4.980
2021-12-11T06:45:00+01:00
4.968
4.968
45733325
1.276
1.276
4.957
6423.134
4.957
2021-12-30T17:21:50Z
4.946
4.946
4.935
4.935
4.924
4.924
2021-12-11T06:40:00+01:00
4.912
4.912
4.901
4.901
45733324
4.890
4.890
1.285
1.285
6423.131
2021-12-30T17:21:42Z
```
Ik snap niet zo goed waarom ook oude berichten lijk te krijgen? Het resultaat is dat de PowerGauge in HomeAssistant heen en weer stuitert tijdens de updates

dennissiemensma, 2021-12-30:
Bedankt voor je melding. Ik heb wat meer context nodig om je te kunnen helpen. He je toevallig ook de topics erbij? Als alternatief kun je voor dsmrreader DEBUG logging inschakelen. Je ziet dan exact wat er gebeurt qua mqtt.

raymonvdm, 2021-12-31:
```
root@mqtt01:/home/raymon# mosquitto_sub -h localhost -t "dsmr/#" -u "user" -P "password"
0.000
3434.664
2021-12-31T09:40:00+01:00
{"read_at": "2020-07-02T23:55:00+02:00", "delivered": "1906.716", "currently_delivered": "0.000"}
{"electricity1": "0.000", "electricity2": "0.000", "electricity1_returned": "0.000", "electricity2_returned": "0.000", "electricity_merged": "0.000", "electricity_returned_merged": "0.000", "electricity1_cost": "0.00", "electricity2_cost": "0.00", "electricity_cost_merged": "0.00", "total_cost": "0.00", "gas": "0.009", "gas_cost": "0.00", "energy_supplier_price_electricity_delivered_1": 0, "energy_supplier_price_electricity_delivered_2": 0, "energy_supplier_price_electricity_returned_1": 0, "energy_supplier_price_electricity_returned_2": 0, "energy_supplier_price_gas": 0}
7.018
0.00
0.000
2.487
0.00
0.000
0.00
9.505
0.000
0
0
0
0
0
0.866
0.00
0.00
50
2
1
4
91
1
0
1.225
0.000
6468.143
6431.610
0.004
0.000
3434.664
1.225
0.000
5
231.0
45741936
2021-12-31T08:40:00Z
2021-12-31T08:42:09Z
45741937
1.243
1.243
6431.614
2021-12-31T08:42:19Z
9.526
2.508
```

dennissiemensma, 2021-12-31:
Bedankt voor je aanvulling. Ik zoek echter de mqtt topics. Ik vermoed namelijk dat die oude datumtijd je gasmeter is, gezien er maar 3 velden in staan. Maar dat blijft een aanname. Je kunt DEBUG logging voor dsmrreader aanzetten of de mqtt client in debug output draaien met -d voor meer informatie: https://github.com/dsmrreader/dsmr-reader/issues/1433#issuecomment-896182322.

raymonvdm, 2021-12-31:
```
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/meter-stats/voltage_swell_count_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/meter-stats/voltage_swell_count_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/meter-stats/voltage_sag_count_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/meter-stats/voltage_sag_count_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/id', ... (8 bytes))
45742406
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_power_current_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_power_current_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_voltage_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_voltage_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_currently_returned_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_currently_returned_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_currently_delivered_l3', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_currently_delivered_l2', ... (0 bytes))
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/phase_currently_delivered_l1', ... (5 bytes))
4.022
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/electricity_currently_delivered', ... (5 bytes))
4.022
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/electricity_delivered_2', ... (8 bytes))
6432.804
Client (null) received PUBLISH (d0, q0, r0, m0, 'dsmr/reading/timestamp', ... (20 bytes))
2021-12-31T09:32:34Z
```

raymonvdm, 2021-12-31:
[dsmr_backend.log](https://github.com/dsmrreader/dsmr-reader/files/7795816/dsmr_backend.log)

dennissiemensma, 2021-12-31:
Hmm. Ik weet dan niet precies hoe je de output krijgt zoals in dat gelinkte ticket. Maar de logging van DSMR-reader zelf is afdoende.

----

## Gasmeter
```
2021-12-31 10:37:45,280 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#2815484) for dsmr/reading/extra_device_timestamp: 2021-12-31T09:35:00Z

2021-12-31 10:37:45,280
2021-12-31T09:35:00Z (UTC)
```

Je gasmeter update om de 5 minuten, dus dat lijkt goed, als in dat het "gat" niet groter is dan dat.

----

## Slimme meter
```
2021-12-31 10:37:45,372 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#2815474) for dsmr/reading/timestamp: 2021-12-31T09:38:01Z

2021-12-31 10:37:45,372
2021-12-31T09:38:01Z (UTC)
```

```
2021-12-31 10:37:55,664 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#2815507) for dsmr/reading/timestamp: 2021-12-31T09:38:11Z

2021-12-31 10:37:55,664
2021-12-31T09:38:11Z (UTC)
```

```
2021-12-31 10:38:00,965 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#2815524) for dsmr/reading/timestamp: 2021-12-31T09:38:17Z

2021-12-31 10:38:00,965
2021-12-31T09:38:17Z (UTC)
```

Je slimme meter lijkt zo'n 16 seconden voor te lopen, maar dat is vrij normaal en maakt verder ook niet uit. De tijdstippen zijn wel consequent met die interval bedoel ik daarmee en dat is goed.

----

- Ik zie alleen zo 1-2-3 bovenstaande datumtijden en **niet die hele oude** (of ik kijk er overheen). Dus je kunt nog kijken of je kunt achterhalen welke topics het zijn of je debuglogging wat langer te laten lopen totdat ze misschien wel voorbij komen.
- Je kunt ook nog overwegen om bij split-topic MQTT config in DSMR-reader alle regels/topics weg te halen die je niet gebruikt. Dat scheelt een hoop ruis bij debuggen en in het algemeen. Je kunt ze immers op elk moment weer inschakelen.

raymonvdm, 2021-12-31:
Misschien kwamen die oude topics dan omdat DSMR reader achter liep? Bij /about stond daarover wel iets, naar nu staat er netjes   

Everything seems to run smoothly. Any issues, such as missing data or stale processes, will be reported here.

dennissiemensma, 2021-12-31:
Ja dat zou kunnen, ik ben je vergeten te vragen of er uberhaupt meldingen waren. Dat was eigenlijk stap 1. Begrijp het goed dat je nu geen afwijkende data meer ziet dan?

dennissiemensma, 2022-01-04:
@raymonvdm Begrijp ik het goed dat je nu geen afwijkende data meer ziet dan?

raymonvdm, 2022-01-04:
Ik had de afgelopen dagen wat moeite met de P1 adapter maar vandaag een nieuwe geplaatst en alles lijkt voor nu okay

dennissiemensma, 2022-01-04:
Top, mocht er toch nog iets vreemds naar voren komen, laat maar weten

closed 2022-01-04 by dennissiemensma