dsmrreader/dsmr-reader #1546: 🙋 Geen hoog tarief meer?

BurgerhoutJ, 2022-01-01:
### Description

Sinds vandaag zie ik alleen maar laag tarief terug in de monitoring. 
Is dit een klein jaarwisseling bugje?
[image]

### DSMR-reader version

4.19

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v4.19 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                              34389
    Consumption records electricity / gas (est.)                                                    5603 / 51
```

dennissiemensma, 2022-01-01:
Als het goed is draait er in de weekenden of op feestdagen geen hoogtarief. Of bedoel je dat niet?

BurgerhoutJ, 2022-01-01:
Oh ja... Het is zaterdag. Sorry. 😏

closed 2022-01-04 by dennissiemensma