dsmrreader/dsmr-reader #1548: OneDrive integratie💡

procmac, 2022-01-05:
### Feature

Is het misschien mogelijk om, naast de DropBox integratie, DSMR-Reader de mogelijkheid te geven om te integreren met OneDrive?

Ik heb  de indruk dat vrij veel gebruikers beschikken over een (betaalde) Microsoft 365 abonnement en daarmee ook beschikken over een ruime OneDrive. DropBox werkt weliswaar óók prettig, maar is nogal beperkt in ruimte als je niet beschikt over een betaalde versie en een betaalde versie ligt wat minder voor de hand als je al beschikt over een (betaalde) ruime Microsoft OneDrive cloud opslag.

dennissiemensma, 2022-01-06:
Hey bedankt voor je verzoek. Ik denk dat het uiteindelijk beter is dat dit, en daarmee ook Dropbox, niet via dsmrreader zelf gaat. Het was een noodgreep destijds ivm sd kaart corruptie en een laatste kans op mogelijk gegevens terugzetten. Achteraf gezien zit er zoveel onderhoud aan dat ik liever tijd besteed aan dsmrreader zelf dan aan dit soort backups. Het kan best zijn dat ik tzt een script of iets aanbied ervoor, maar dit is zo generiek dat het in principe uiteindelijk via elke tool of implementatie ervan zou kunnen die bij de backups van dsmrreader kan.

procmac, 2022-01-06:
Jammer, maar begrijpelijk. Bedankt voor je antwoord.

closed 2022-01-06 by procmac