dsmrreader/dsmr-reader #1549: 🙋

Donmacha, 2022-01-05:
### Description

Hallo, ik probeer dsmrreader te installeren maar  als ik ./manage.py migrate invoer gaat het verkeerd.
Wat doe ik mis of ben ik vergeten ?

mvg Dries

bord: odroid xu4
os op emmc
os : armbian

### DSMR-reader version

v4

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
<!-- COPY OUTPUT BELOW -->

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?
could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_debuginfo.py", line 33, in handle
    self._dump_application_info()
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_debuginfo.py", line 46, in _dump_application_info
    for line in self._intercept_command_stdout('showmigrations', no_color=True).split("\n"):
  File "/home/dsmr/dsmr-reader/dsmr_backend/tests/mixins.py", line 12, in _intercept_command_stdout
    call_command(command, stdout=stdout, stderr=stderr, *args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 168, in call_command
    return command.execute(*args, **defaults)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/commands/showmigrations.py", line 52, in handle
    return self.show_list(connection, options['app_label'])
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/commands/showmigrations.py", line 71, in show_list
    loader = MigrationLoader(connection, ignore_no_migrations=True)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/migrations/loader.py", line 53, in __init__
    self.build_graph()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/migrations/loader.py", line 216, in build_graph
    self.applied_migrations = recorder.applied_migrations()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/migrations/recorder.py", line 77, in applied_migrations
    if self.has_table():
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/migrations/recorder.py", line 55, in has_table
    with self.connection.cursor() as cursor:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?
could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?
```

dennissiemensma, 2022-01-06:
Bedankt voor je vraag. Het lijkt erop dat de database niet draait of niet bereikbaar is. Heb je die wel gestart/cluster aangemaakt? Afhankelijk van het OS gebeurt dat wel of niet automatisch. Mogelijk mist de cluster nog, zie de blauwe notitie in https://dsmr-reader.readthedocs.io/nl/v4/tutorial/installation/step-by-step.html#database-backend-postgresql. Let op dat die versienummers nog vrij oud zijn. Wellicht dat je postgres versie nu 12 of 13 oid is.

dennissiemensma, 2022-01-06:
Bij nader inzien had je anders ook al een fout gehad bij het aanmaken van de FB user en database zelf. Kun je je herinneren of dat wel goed ging?

Donmacha, 2022-01-06:
```
odroid@odroidxu4:/$ pg_createcluster 12 main --start
Error: cluster configuration already exists
```
die lijkt goed, de database lijkt mij ook correct aangemaakt:
```
odroid@odroidxu4:/$ sudo -u postgres psql
psql (12.9 (Ubuntu 12.9-0ubuntu0.20.04.1))
Type "help" for help.

postgres=# /l
postgres-# \l
                                   List of databases
    Name    |   Owner    | Encoding |   Collate   |    Ctype    |   Access privileges
------------+------------+----------+-------------+-------------+-----------------------
 dsmrreader | dsmrreader | UTF8     | en_US.UTF-8 | en_US.UTF-8 |
 postgres   | postgres   | UTF8     | en_US.UTF-8 | en_US.UTF-8 |
 template0  | postgres   | UTF8     | en_US.UTF-8 | en_US.UTF-8 | =c/postgres          +
            |            |          |             |             | postgres=CTc/postgres
 template1  | postgres   | UTF8     | en_US.UTF-8 | en_US.UTF-8 | =c/postgres          +
            |            |          |             |             | postgres=CTc/postgres
(4 rows)

postgres-#
```

dennissiemensma, 2022-01-06:
Ik denk dat je dan moet achterhalen op welke poort de DB draait. Dit zou de standaard moeten zijn:
```
Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?
```

Je kunt kijken met bijvoorbeeld netstat:
```
sudo netstat -plant | grep postgres
```

Bij mij:
```
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      636/postgres        
tcp6       0      0 ::1:5432                :::*                    LISTEN      636/postgres        
tcp6       0      0 ::1:5432                ::1:57222               ESTABLISHED 7070/postgres: 11/m 
tcp6       0      0 ::1:5432                ::1:57224               ESTABLISHED 7072/postgres: 11/m 
tcp6       0      0 ::1:5432                ::1:46964               ESTABLISHED 1649/postgres: 11/m 
tcp6       0      0 ::1:5432                ::1:58334               ESTABLISHED 20070/postgres: 11/ 
tcp6       0      0 ::1:5432                ::1:57220               ESTABLISHED 7069/postgres: 11/m 
tcp6       0      0 ::1:5432                ::1:58352               ESTABLISHED 20081/postgres: 11/ 
```

Donmacha, 2022-01-07:
ok is gelukt ,

poort van postgres was standaard 5433 , had daar overgekeken.
bedankt voor de hulp !

mvg, dries

closed 2022-01-07 by Donmacha