dsmrreader/dsmr-reader #1550: 🚨 Docker container user heeft niet genoeg rechten

attaxia, 2022-01-06:
### Description

Drie dagen geleden de docker container geupgrade, verder geen wijzigingen. Sindsdien geen statistieken meer.
Logs staan vol met:

`RuntimeError: ('Failed to connect: {}', SerialException(13, "could not open port /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AI2BAI4B-if00-port0: [Errno 13] Permission denied: '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AI2BAI4B-if00-port0'"))
`

Ik heb het probleem op kunnen lossen door het volgende commando (in de container) uit te voeren:

### DSMR-reader version

4.19

dennissiemensma, 2022-01-06:
Je kunt het beste hiervoor een issue aanmaken in het docker project van xirixiz. Vh5eck ook even of er wellicht anderen zijn die eerder hetzelfde gemeld hebben.

dennissiemensma, 2022-01-06:
Zie mogelijk https://github.com/xirixiz/dsmr-reader-docker/issues/251

closed 2022-01-06 by dennissiemensma

poerlemans, 2022-01-07:
The development release of the docker image looks solving the problem.