dsmrreader/dsmr-reader #1558: 🚨

Janoz-NL, 2022-01-16:
### Description

Ik laat mijn dsmr-reader zijn gegevens via MQTT doorsturen naar Home Assistant.

Na een week lijkt het alsof er geen MQTT berichten meer doorgestuurd worden. De grafieken in HA houden abrupt op. In DSMR-Reader zelf staat nog wel data, en ook het wegschrijven naar InfluxDB blijft werken.

Zodra ik de MQTT broker/connection disable en dan weer enable werkt het weer. 

Ik heb zelf even gekeken naar wat er op de queue gezet wordt en het lijkt er op dat hij ALLEEn de ID opstuurt. Dus zeg maar alleen de eerste regel van het bericht

`dsmr/reading/id 25459933
dsmr/reading/id 25550983
dsmr/reading/id 25561729
dsmr/reading/id 25479906
dsmr/reading/id 25501059
dsmr/reading/id 25510984`

Na herstarten zitten alle tussenliggende regels er wel weer tussen

`dsmr/reading/id 26348050
dsmr/reading/phase_voltage_l3 236.0
dsmr/reading/phase_currently_delivered_l3 0.324
dsmr/reading/phase_currently_delivered_l2 0.456
dsmr/reading/phase_currently_delivered_l1 0.046
dsmr/reading/electricity_currently_delivered 0.827
dsmr/reading/timestamp 2022-01-16T15:00:06Z
dsmr/reading/id 26348051
dsmr/reading/phase_voltage_l3 235.0
dsmr/reading/phase_currently_delivered_l3 0.301
dsmr/reading/phase_currently_delivered_l2 0.470
dsmr/reading/phase_currently_delivered_l1 0.047
dsmr/reading/electricity_currently_delivered 0.820
dsmr/reading/electricity_delivered_1 11327.338
dsmr/reading/timestamp 2022-01-16T15:00:07Z`

(newlines vallen weg in de berichtvoorbeelden)

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-01-16:
Bedankt voor je melding. Je kunt het beste DEBUG-logging inschakelen als dit vaker gebeurt. Je zou dan exact moeten zien wat DSMR-reader doet, queue'd en verstuurt.

- https://dsmr-reader.readthedocs.io/nl/latest/how-to/troubleshooting/enabling-debug-logging.html

Waren er op dat moment geen andere fouten of waarschuwingen in DSMR-reader?

Janoz-NL, 2022-01-16:
Hmm, ik heb geprobeerd de enviroment variable toe te voegen, maar nu wil mijn dockerimage niet meer starten:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 10-adduser: executing... 
usermod: no changes

-------------------------
User UID: 803
User GID: 803
-------------------------

[cont-init.d] 10-adduser: exited 0.
[cont-init.d] 20-set-app-defaults: executing... 
[ INFO ] DSMR Reader release: 4.19.0
[ INFO ] DSMR Reader Docker release: 2022.01.04
[ INFO ] Ensure permissions...
[ INFO ] Setting architecture requirements...
[ INFO ] X64 Architecture
[ INFO ] Verifying if the DSMR web credential variables have been set...
[ INFO ] Verifying database connectivity to host: postgres with port: 5432...
[ INFO ] Database connectivity successfully verified!
[ INFO ] Running post configuration...
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_dropbox, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.
0 static files copied to '/var/www/dsmrreader/static', 563 unmodified.
Updating password of superuser "admin"
Deactivating any other existing superusers
[ INFO ] Checking for NGINX SSL configuration...
[ INFO ] ENABLE_NGINX_SSL is disabled, nothing to see here. Continuing...
[ INFO ] Checking for HTTP AUTHENTICATION configuration...
[ INFO ] ENABLE_HTTP_AUTH is disabled, nothing to see here. Continuing...
[ INFO ] Configuring DSMR in receiver datalogger mode....
[cont-init.d] 20-set-app-defaults: exited 0.
[cont-init.d] done.
[services.d] starting services
Starting DSMR Reader - nginx...
Starting DSMR Reader - webinterface...
Starting DSMR Reader - backend...
[services.d] done.
/bin/bash: line 1: /app/run.sh: No such file or directory
d] /bin/bash exited 127
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

Janoz-NL, 2022-01-16:
Ik heb debugging aangezet. Over een week weet ik hopelijk meer

Janoz-NL, 2022-01-21:
Zo te zien lijkt het nu weer te gebeuren. Na verloop van tijd komen er steeds minder meetresultaten binnen. Ik zie nu alleen id's voorbij komen op het mqtt kanaal. Het lijkt er op dat hij alles afkeurt omdat het 'hetzelfde is als het vorige bericht'. Mogelijk zit daar iets niet goed. 

[log.txt](https://github.com/dsmrreader/dsmr-reader/files/7915362/log.txt)

Dit is het log. en hoewel daar staat dat sommige berichten wel doorkomen zie ik alleen maar ID's op het kanaal en heel af en toe phase_voltage_l3 en extra_device_timestamp en phase_currently_delivered_l1

dennissiemensma, 2022-01-21:
Bedankt voor je update. Dit verklaart een hoop. 

Wat er gebeurt is dat DSMR-reader alle MQTT-berichten met een `retain` vlag stuurt. Dat betekent dat de broker altijd de laatste waarde per topic bewaart, ook als er pas later nieuwe (of dezelfde client) verbindt. Echter kan het zijn dat dat niet lekker werkt als de client daar niet mee om kan gaan. 

https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_mqtt/services/messages.py#L33-L38

Er is geen makkelijke instelling om dit uit te zetten, maar in principe kun je het overschrijven door in `dsmrreader/settings.py` dit toe te voegen en de applicatie te herstarten:
```
# dsmrreader/settings.py
CACHES['mqtt']['TIMEOUT'] = 0
```

Er staat waarschijnlijk al een andere waar je die bij kan zetten:
```
# dsmrreader/settings.py
CACHES['default']['TIMEOUT'] = 60
CACHES['mqtt']['TIMEOUT'] = 0
```

Alleen kan het zijn dat het in Docker wat lastiger is. Ik zal voor een volgende versie er een env var voor maken.

dennissiemensma, 2022-01-21:
BIj nader inzien, zie https://github.com/dsmrreader/dsmr-reader/blob/v4/docs/reference/env-settings.rst#dsmrreader-mqtt-max-cache-timeout