dsmrreader/dsmr-reader #1562: 🤔 DSMR reader compatible with 64bits Raspbian OS?

mkruiver, 2022-01-22:
### Your question

Is DSMR reader eigenlijk compatible met 64 bits Raspbian?

https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/

of zouden misschien zelfs de Python packages etc nog niet beschikbaar zijn voor 64bits?

mkruiver, 2022-01-22:
@dennissiemensma al eens geprobeerd?

mkruiver, 2022-01-22:
Raspbian 64 lijkt iig met Python 3.9 voor ARM64 te komen.
Postgres is ook ARM64 compatible.

mkruiver, 2022-01-23:
Werkt gewoon! :)

closed 2022-01-23 by mkruiver

dennissiemensma, 2022-01-23:
@mkruiver bedankt voor de check! Ik dacht dat alles inmiddels al 64 bit was, maar fijn dat het werkt!

mkruiver, 2022-02-03:
Hoppa, Raspbian 64bits OS is officieel uit en DSMRReader draait al op 64bits ;)

https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/