dsmrreader/dsmr-reader #1563: openapi specificatie json file bevat yaml🚨

mvaneijken, 2022-01-26:
### Description

Als ik en DSMR-Reader de /api/v2/openapi.json download, dan bevat deze de specificatie in yaml formaat ipv json. Nu weet ik dat yaml en json op zich uitwisselbaar zijn, maar de extensie suggereert dat het om json gaat.

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-01-26:
Bedankt voor je melding! Dit is niet bewust en ook niet bedoeld. Blijkbaar kan ReDoc met beide formaten overweg, want zo is niet mijn insteek. Ik zal de extensie aanpassen in DSMR-reader v5.x

dennissiemensma, 2022-01-26:
@mvaneijken het was wat uitzoekwerk hoe het onderwater zit in dat API-framework. Bij jezelf er JSON uit te trekken kun je dit er achter zetten `?format=openapi-json`:
```
GET /api/v2/openapi.json?format=openapi-json
```

In DSMR-reader v5 zal ik de bestandsextensie weghalen. Gezien het dynamisch is.

dennissiemensma, 2022-01-26:
Aangepast in 904e41400d59a3992f4ce7d840ec697840705a7d

closed 2022-01-26 by dennissiemensma

mvaneijken, 2022-01-26:
Dankje voor je reactie en uitzoeken! Ik ga het proberen met de genoemde parameter.

commit 5f9d6daa (2022-01-26): Redoc QoL fixes #1563