dsmrreader/dsmr-reader #1568: 🙋

karia001, 2022-01-31:
### Description

influxdb 2.0 intergratie werkt niet gezien de token 89 karakters is en eindigt op ==  en deze hoeveelheid karakters niet wordt geaccepteerd

### DSMR-reader version

5

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                     v5.0rc1 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             422795
    Consumption records electricity / gas (est.)                                                61777 / 29501

UNRESOLVED ISSUES
    Too many outgoing InfluxDB measurements queued for transit                          4 months, 3 weeks ago
```

dennissiemensma, 2022-01-31:
Bedankt voor het proberen en de melding! Ik zal zo de v5-branch bijwerken met een update die de velden op 128 tekens max zet.

dennissiemensma, 2022-01-31:
@karia001 je kunt proberen te updaten naar release candidate # 2. Als het goed is zit er een DB-migratie bij die de velden (org, bucket, api-token) alledrie 128 tekens lang toestaat.

4c35b569228444d01e67bc276edfcc67aee4c3b9

dennissiemensma, 2022-01-31:
https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.0.0-rc2

dennissiemensma, 2022-01-31:
Laat maar weten of dat het oplost en of er andere andere zaken naar voren komen!

karia001, 2022-01-31:
Hoi Dennis,

ja het is opgelost, bedankt voor het snel schakelen:-)

closed 2022-01-31 by karia001

dennissiemensma, 2022-02-04:
@karia001 hoe werkt de nieuwe integratie naar InfluxDB 2.x verder? Komt alle data nog steeds goed binnen?

karia001, 2022-02-04:
Dennis, 

bij mij draait het tot nu toe goed!

dennissiemensma, 2022-02-04:
Top, fijn om te horen. Dank voor je vlotte reactie!