dsmrreader/dsmr-reader #1574: OpenWeather API integratie💡

floyson-reference, 2022-02-06:
### Feature

Hi,

Zijn er toevallig al plannen om ooit OpenWeather api (https://openweathermap.org/current) te integreren. Ik zie dat nu buienradar.nl wordt gebruikt, maar voor België werkt dit helaas niet.

dennissiemensma, 2022-02-06:
Bedankt voor je verzoek! Ik verwacht dat de weer-integratie uiteindelijk uit DSMR-reader gaat. Achteraf gezien had ik het er niet in moeten opnemen.

Afhankelijk van je behoefte kun je overwegen om bijvoorbeeld je gasstanden te exporteren naar MinderGas, of via een andere tool temperatuur bij te houden.

closed 2022-02-06 by dennissiemensma

bertusdebruin, 2022-02-14:
@dennissiemensma Ik hoop dat de huidige weer-integratie uiteindelijk blijft.
Het weer is tegenwoordig erg wisselvallig.

Het is voor mij ideaal om last minute veranderingen in het verbruik in een oogopslag waar te nemen en te verklaren.
Voorbeeld: Afgelopen nacht begon het weer te vriezen en sloeg de CV-ketel snachts pardoes aan.
Deze huidige nacht is het relatief warm, rond de 10 graden en dat verklaart het verbruik.

Met externe oplossingen als MinderGas ben ik minder blij.
Ik gebruik niet voor niets DSMR-reader :) waarbij ik lokaal mijn gegevens opsla zonder cloud of kekke APP.

Anders had ik net zo goed een kant en klare P1 meter kunnen kopen.
Waarbij mijn verbruik continu naar derden worden geupload.

dennissiemensma, 2022-02-14:
@bertusdebruin bedankt voor je aanvulling. Er is zeker wat voor te zeggen, alleen weet ik niet of DSMR-reader daar de plek voor is. 

In de basis is het project alleen het uitlezen van de slimme meter en het plotten van "een" visualisatie. Daarnaast spelen graaddagen ook mee (https://www.mindergas.nl/degree_days/explanation) voor een goede berekening qua temperatuur. Nou snap ik de overweging om wel of niet je verbruik naar MinderGas (of een andere partij) te sturen, maar dan zit je nog steeds met dat je eigenlijk niet alle factoren meeneemt.
En als je alleen een globale historie wilt van temperatuurmetingen, dan zijn er vast wel tools of integraties die het lokaal voor je opslaan. Daarom is bijvoorbeeld HA wellicht wel een optie, al weet ik niet exact hoe/waar deze gegevens opgeslagen worden. Bijvoorbeeld bij: https://www.home-assistant.io/integrations/openweathermap/

Ik wil eigenlijk toe naar een wat beter te onderhouden project dat vooral bij de kern ligt van het uitlezen van de slimme meters. Het is ook veel makkelijker om een tool te onderhouden die 1 ding goed wilt doen, dan een tool die van alles een beetje doet. En dat merkte ik al de afgelopen jaren bij de keuzes die ik eerder gemaakt heb en ik tegenwoordig zeker niet (of minder snel) zal maken. Vandaar.