dsmrreader/dsmr-reader #1575: 🙋 Upgrade rc2 naar 5.0 geeft fout?

Calimerorulez, 2022-02-08:
### Description

Hi,

Ik zit op rc2, en probeer de laatste sources te 'pullen', maar ik krijg onderstaande error:

```
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
```

En dan met name de laatste regel. Zie ik iets over het hoofd? :)

### DSMR-reader version

5.0-rc2

dennissiemensma, 2022-02-08:
Bedankt voor je melding. Ik ken deze fout niet, ik ken de termen wel. Als ik hier wat lees dan denk ik dat er een keuze gemaakt moet worden "hoe te mergen": https://stackoverflow.com/questions/62653114/how-can-i-deal-with-this-git-warning-pulling-without-specifying-how-to-reconci

Alleen snap ik niet waarom het opeens een issue is, omdat je (als het goed is) zelf niets commit lokaal en alleen "een kopie trekt" van dit project. 

Wat is de output van:
```
sudo su - dsmr

git fetch
git branch
```

Calimerorulez, 2022-02-08:
> Bedankt voor je melding. Ik ken deze fout niet, ik ken de termen wel. Als ik hier wat lees dan denk ik dat er een keuze gemaakt moet worden "hoe te mergen": https://stackoverflow.com/questions/62653114/how-can-i-deal-with-this-git-warning-pulling-without-specifying-how-to-reconci
> 
> Alleen snap ik niet waarom het opeens een issue is, omdat je (als het goed is) zelf niets commit lokaal en alleen "een kopie trekt" van dit project.
> 
> Wat is de output van:
> 
> ```
> sudo su - dsmr
> 
> git fetch
> git branch
> git status
> ```

```
root@dsmr-5:~# sudo su - dsmr
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git fetch
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 10 (delta 5), reused 8 (delta 3), pack-reused 0
Unpacking objects: 100% (10/10), 1.26 KiB | 76.00 KiB/s, done.
From https://github.com/dsmrreader/dsmr-reader
   96b23e39..01e78996  v5         -> origin/v5
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
 ! [new branch]        wip/v5.1   -> origin/wip/v5.1  (unable to update local ref)
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git branch
  v4
* v5
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git status
On branch v5
Your branch is behind 'origin/v5' by 13 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        .env.save
        .env.save.1

nothing added to commit but untracked files present (use "git add" to track)
(.venv) dsmr@dsmr-5:~/dsmr-reader$ 
```

Calimerorulez, 2022-02-08:
Ik wil ook wel `refs/remotes/origin/wip/v5.1` proberen te verwijderen lokaal en kijken wat ie dan doet?

dennissiemensma, 2022-02-08:
Er van uitgaande dat je deze niet bewust hebt gemaakt of ze niet wilt bewaren: 
```
        .env.save
        .env.save.1
```
(en anders even backuppen voordat je onderstaande doet)

Je kunt dit proberen:
```
sudo su - dsmr
git reset HEAD --hard
```

Nu zou dit andere output moeten geven:
```
git status
```

Als iets: `nothing to commit, working tree clean`

Zo ja, probeer dan nogmaals:
```
git pull
```

Calimerorulez, 2022-02-08:
```
root@dsmr-5:~# sudo su - dsmr
(.venv) dsmr@dsmr-5:~/dsmr-reader$ rm .env.save   
(.venv) dsmr@dsmr-5:~/dsmr-reader$ rm .env.save.1 
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git reset HEAD --hard
HEAD is now at 41619764 Dependency updates
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git status
On branch v5
Your branch is behind 'origin/v5' by 13 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

nothing to commit, working tree clean
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
From https://github.com/dsmrreader/dsmr-reader
 ! [new branch]        wip/v5.1   -> origin/wip/v5.1  (unable to update local ref)
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git branch -d wip/v5.1
error: branch 'wip/v5.1' not found.
(.venv) dsmr@dsmr-5:~/dsmr-reader$
```

Branch wip/5.1 bestaat niet...

dennissiemensma, 2022-02-08:
Ik denk dat er twee verschillende dingen spelen. Wellicht heb je een nieuwere Git versie.

Probeer dan dit:
```
git config pull.ff only

git pull
```

Fast-Forward zou namelijk goed moeten gaan, gezien de melding eerder:
```
Your branch is behind 'origin/v5' by 13 commits, and can be fast-forwarded.
```

Calimerorulez, 2022-02-08:
```
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git config pull.ff only
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git pull
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
From https://github.com/dsmrreader/dsmr-reader
 ! [new branch]        wip/v5.1   -> origin/wip/v5.1  (unable to update local ref)
(.venv) dsmr@dsmr-5:~/dsmr-reader$ 
```

Calimerorulez, 2022-02-08:
Ik wil ook gewoon opnieuw beginnen hoor, ik maak een backup van de db en begin met een schone container, geen probleem :)

dennissiemensma, 2022-02-08:
Ik denk dat je uiteindelijk tegen hetzelfde aanloopt. Ik zal even verder zoeken, want het is vreemd dat die stuk gaat op `wip/v5.1/1567/Justin991q-dark-mode`. Dat was een branch die ik korte tijd had gemaakt.

Calimerorulez, 2022-02-08:
Geen haast, alvast bedankt voor je hulp!

dennissiemensma, 2022-02-08:
Kun je deze eens proberen?
```
git gc --prune=now
```
https://stackoverflow.com/a/27278221

En daarna nog eens `git pull`

Calimerorulez, 2022-02-08:
```

(.venv) dsmr@dsmr-5:~/dsmr-reader$ git gc --prune=now
Enumerating objects: 33315, done.
Counting objects: 100% (33315/33315), done.
Compressing objects: 100% (9938/9938), done.
Writing objects: 100% (33315/33315), done.
Total 33315 (delta 20948), reused 33235 (delta 20893), pack-reused 0
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git pull
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
From https://github.com/dsmrreader/dsmr-reader
 ! [new branch]        wip/v5.1   -> origin/wip/v5.1  (unable to update local ref)
(.venv) dsmr@dsmr-5:~/dsmr-reader$ 
```

dennissiemensma, 2022-02-08:
En deze staat daar ook nog verder onderin:
```
git remote prune origin
```

Calimerorulez, 2022-02-08:
Yes!

```
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git remote prune origin
Pruning origin
URL: https://github.com/dsmrreader/dsmr-reader.git
 * [pruned] origin/wip/dep-check-fix
 * [pruned] origin/wip/v4.20
 * [pruned] origin/wip/v5.1/1567/Justin991q-dark-mode
(.venv) dsmr@dsmr-5:~/dsmr-reader$ git pull
From https://github.com/dsmrreader/dsmr-reader
 * [new branch]        wip/v5.1   -> origin/wip/v5.1
Updating 41619764..71d8c78f
Fast-forward
 .github/ISSUE_TEMPLATE/config.yml             | 10 ++++++++++
 .github/workflows/automated-tests.yml         | 18 ++----------------
 .github/workflows/dependency-update-check.yml |  8 ++++----
 README.md                                     | 31 +++++++++++++++----------------
 docs/reference/changelog.rst                  |  4 ++--
 dsmrreader/__init__.py                        |  2 +-
 dsmrreader/provisioning/downgrade/v5.0.0.sh   |  2 +-
 7 files changed, 35 insertions(+), 40 deletions(-)
(.venv) dsmr@dsmr-5:~/dsmr-reader$
```

dennissiemensma, 2022-02-08:
Ah kijk, ik krijg nu de fout zelf ook:

```
(.venv) dsmr@rpi4:~/dsmr-reader $ git fetch
remote: Enumerating objects: 345, done.
remote: Counting objects: 100% (233/233), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 137 (delta 95), reused 132 (delta 90), pack-reused 0
Receiving objects: 100% (137/137), 33.73 KiB | 4.82 MiB/s, done.
Resolving deltas: 100% (95/95), completed with 48 local objects.
From https://github.com/dsmrreader/dsmr-reader
   96b23e39..71d8c78f  v5                     -> origin/v5
 * [new branch]        feature/1567/dark-mode -> origin/feature/1567/dark-mode
   8048d67a..7c9f4dba  v4                     -> origin/v4
error: cannot lock ref 'refs/remotes/origin/wip/v5.1': 'refs/remotes/origin/wip/v5.1/1567/Justin991q-dark-mode' exists; cannot create 'refs/remotes/origin/wip/v5.1'
 ! [new branch]        wip/v5.1               -> origin/wip/v5.1  (unable to update local ref)
 * [new tag]           v4.20.0                -> v4.20.0
 * [new tag]           v5.0.0                 -> v5.0.0
error: some local refs could not be updated; try running
 'git remote prune origin' to remove any old, conflicting branches
```

dennissiemensma, 2022-02-08:
Ja juist, dus daar staat dan wel expliciet:
```
 'git remote prune origin' to remove any old, conflicting branches
```

Nja, mooi!

Calimerorulez, 2022-02-08:
Dan ga ik nu verder met de upgrade ;)
Dank je!

closed 2022-02-08 by Calimerorulez

dennissiemensma, 2022-02-08:
Ik zal dit issues pinnen. Met een beetje pech krijgt iedereen hier last van namelijk.

dennissiemensma, 2022-02-08:
Ik denk dat ik ooit eerder een v5.1 branch gemaakt hebt, die verwijderd, en nu v5.0 er is, weer opnieuw 5.1 maak. Maar ondertussen is die eerste versie van de branch wel her en der binnengehaald en dat geeft issues.

Ik zal deze verwijderen en een andere naam gebruiken. Dat fixt het vermoed ik.

Calimerorulez, 2022-02-08:
De rest van de upgrade-procedure (ik heb die van v4 naar v5 deels gevolgd om van rc2 naar 5 te gaan) gaat verder wel goed, hij draait als een zonnetje :)