dsmrreader/dsmr-reader #1577: psycopg2-binary geeft problemen bij upgrade naar 5.0🚨

dajappie, 2022-02-09:
### Description

Hi Dennis,

Bij het upgraden van 4.20 naar 5.0 heb ik ook Python geupgrade van 3.9.2 naar 3.9.9 en wat cleanup gedaan van oude Python 2.7, 3.5 en 3.6 versies. Postgres is gelijk gebleven op 9.6.24. Nu is de requirement voor psycopg2 blijkbaar een lastige. Als ik gebruik maak van psycopg2-binary (momenteel v2.9.3, maar ook issue op oudere versies zoals 2.8.4)  krijg ik een fout "undefined symbol PQencryptPasswordConn".  Pas na een non-binary install via "pip3 install --no-binary :all: psycopg2" waarbij opnieuw gecompileerd werd werkte deze naar behoren. Nu het probleem: bij een deploy.sh wordt ook uit ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt toch weer psycopg2-binary geinstalleerd, waardoor de interactie met Postgres niet meer werkt. Pas na uitcommenten van psycopg2-binary in requirements/base.txt, en aanpassen van deploy.sh om de handmatige wijziging van requirements/base.txt te accepteren, kon de installatie voltooid worden. Voor toekomstige updates zal dat echter een hoop handwerk opleveren om elke keer de geupdate base.txt te commenten, de binary install terug te draaien etc. Nu kan ik niet vinden waarop dit misgaat, Python 3.9.9 is van scratch geinstalleerd zoals in de upgrade guide, en Postgres is as-is zoals werkend onder 4.20. Lijkt een foutief gelinkte library, maar kan er niks zinnigs over vinden. Enig idee?

### DSMR-reader version

5.0

### DSMR-reader platform

Native (e.g. manual installation)

dajappie, 2022-02-09:
Kleine toevoeging, blijkbaar is PQencryptPasswordConn pas geintroduceerd in libpq in PostgreSQL 10. Echter op mijn RPi3b met Stretch is de laatste Postgres package 9.6 ook voor de client. Ik vermoed dat alleen een verse, lokaal gecompileerde Postgres 10+ dan gaat werken....

dennissiemensma, 2022-02-09:
Ik denk dat je er niet aan ontkomt om postgres te upgraden. Versie 9 is inmiddels geheel EOL sinds vorig jaar. Ik zou zelf versie 13 pakken, dan kun je even voorruit. Verder zou je nog kunnen kijken naar docker, dan ben je minder van je OS afhankelijk, al heeft Docker ook een leercurve.

dajappie, 2022-02-10:
Klopt, was meer dat in je release notes staat dat Postgres >9.5 moet zijn omdat Django 3.2 compatibel zou moeten zijn met Postgres 9.6. Als ik kijk op https://docs.djangoproject.com/en/3.2/ref/databases/#postgresql-notes zou dat ook zo moeten zijn. Niet dus. Maar eens kijken of we Postgres een bumpje kunnen geven.....

dennissiemensma, 2022-02-10:
De docs worden zo bijgewerkt voor `libopenjp2`, dank!

dennissiemensma, 2022-02-10:
Ik zal trouwens ook de release notes bijwerken zodat PostgreSQL 9 gewoon niet meer ondersteund wordt. En dat de aanbevolen versie 13 is