dsmrreader/dsmr-reader #1578: In de war met de berekening🤔

DavyRoswinkel, 2022-02-10:
### Your question

Mijn Energie leverancier berekend het verbruik op de volgende manier:
In dit voorbeeld 270 kWh
€ 0,24054/kWh x 135 kwh = €32,47
€ 0,22529/kWh x 135 kwh = €30,41
Vaste leveringskosten (€ 5,99/maand) € 0,19692/dag x 31 dagen = 6,10
voor Enexis B.V. : Capaciteitstarief t/m 3 x 25A of t/m 1 x 80A (€ 267,68/jaar) € 0,73338/dag x 31 dagen = 22,73
Vermindering energiebelasting (- € 824,77/jaar) - € 2,25965/dag x 31 dagen = 70.04 euro
Totaal tarief gas (1 t/m 170.000 m3) : € 1,16066/m3 x 21 m3 = 24.37
Vaste leveringskosten (€ 5,99/maand) € 0,19692/dag x 31 dagen = 6.10
voor Enexis B.V. : Capaciteitstarief t/m 10 m3/uur, stand.jaarv. < 500 m3 (€ 138,19/jaar) € 0,37861/dag x 31 dagen = 11.74
Totaal; 42.21

Het maand overzicht dat ik van mijn leverancier krijg:
270 kWh kost €21,15
and
21m3 kost € 42.21
De leverancier neemt dus alle dagelijkse verbruikskosten en de energiebelasting teruggaaf ( deze berekenen ze alleen/meteen over de elektra)

De dsmr reader laat het volgende zien: € 62,88 voor de elektra € 24,37 voor gas
Daarna berekend de dsrm reader de dagelijkse kosten en de belasting teruggaaf ( -€23.25)

De paar centen verschil is het punt niet ( afronding en misschien wat dingen gemist ofzo via dsmr)
De uitkomst is nagenoeg hetzelfde alleen de verbruiks totalen per maand zijn anders.

Ben ik nu de weg kwijt? Of is dit op te lossen. Of heb ik gewoon een energie leverancier die de zaken op een aparte manier berekend?
Ik was een beetje in de war, ik kan de berekeningen van de dsmr reader dromen....maar vandaag ontving ik een email met heel andere cijfers van de energie leverancier

dennissiemensma, 2022-02-10:
Bedankt voor je vraag. Ik denk dat de korte versie is dat je leverancier altijd leidend is en DSMR-reader puur een indicatie.

De lange versie is dat DSMR-reader alleen kosten laat zien om je een gevoel te geven hoeveel een dag je "kost" ten opzichte van andere dagen. Dus hoeveel Euro kost het je als je een dagje 18 graden thermostaat doet in de winters tov 20 graden. Idem voor elektra. Het zal nooit in de buurt komen van wat leveranciers berekenen, omdat er zoveel heffingen en alles zijn. Het enige wat je kan vergelijken zijn de meterstanden. 
Je geeft aan dat die niet overeenkomen. Je zou kunnen kijken welke datum ze pakken en dan in het Archief in DSMR-reader die datum (of de dag ervoor/erna) erbij pakken. Je ziet dan als het goed is ook de eerst gemeten meterstand van de dag. Die zou overeen moeten komen met je specificatie.

Ik had zelf toevallig laatst dat mijn contractduur in DSMR-reader 2 dagen te lang stond, waardoor ik ook een mismatch had. Maar toen ik eenmaal de datums vergeleek en in het Archief keek naar de datums volgens de leverancier, bleek het toch te kloppen. Kan het zijn dat dat ook bij jou speelt?

DavyRoswinkel, 2022-02-10:
Nee ik bedoel het dus anders.
Maar het is moeilijk uit te leggen.

De leverancier zegt dus dat ik in een maand aan elektra bijvoorbeeld 21,15 euro heb verbruikt, maar dit zijn dan niet alleen de verbruikskosten. Maar inclusief de dagelijkse kosten en minus de energie belasting. Ik krijg dan dus een email met: u hebt afgelopen maand 270 kWh verbruikt en dat kost 21,15 euro.
21m3 gas kost 42,21

Via het overzicht in de dsmrreader zie je eerst de verbruikskosten en later gaan de dagelijkse kosten en de energiebelasting er pas af en bij.Van de elektra en het gas.

Het komt door de manier van rekenen van de leverancier, hierdoor lijkt het natuurlijk nog voordeliger. 
Bij mijn vorige leverancier werd precies dezelfde manier van rekenen gebruikt als in de dsmrreader, eerst alle verbruikskosten en later alle vaste kosten en de energiebelasting.

dennissiemensma, 2022-02-10:
Bedankt voor je aanvulling. Ik begrijp je denk ik heel goed en daar doel ik dus op, met dat het niet te vergelijken is. 

Je kunt in DSMR-reader hooguit kijken of die `270 kWh` en `21 m3` gas klopt. Al het andere qua prijzen en berekeningen is onmogelijk te verifieren.

DavyRoswinkel, 2022-02-10:
> Bedankt voor je aanvulling. Ik begrijp je denk ik heel goed en daar doel ik dus op, met dat het niet te vergelijken is. 
> 
> Je kunt in DSMR-reader hooguit kijken of die `270 kWh` en `21 m3` gas klopt. Al het andere qua prijzen en berekeningen is onmogelijk te verifieren.

Het punt is dus dat je juist dat niet kan zien.
In de dsmrreader is de totale prijs van het aantal kWh bruto, dus voor de dagelijkse kosten en de energiebelasting.

Mijn energie leverancier geeft deze prijs alleen netto en telt daarbij dus ook de volledige energiebelasting teruggaaf bij op.( alleen bij de elektra dus)

Helemaal onderaan de streep, zijn de dsmrreader en de energieleverancier gelijk. ( dus gas+elektra +fixed kosten)

De dsmrreader doet het zo : gas( verbruikskosten)+
 elektra(verbruikskosten) + fixedkosten( vaste kosten gas+vaste kosten elektra+energiebelastingteruggaaf)= totaal aan kosten

Mijn energie leverancier doet het zo: 
elektra( verbruikskosten+ vaste kosten + energiebelasting teruggaaf) + gas( verbruikskosten + vaste kosten) = totaal aan kosten

Het enige wat ik duidelijk wilde maken is dat het verschil van reken manier juist zo verwarrend is. Het ligt aan deze leverancier, men kan nu namelijk onmogelijk het juiste verbruik transparant in de gaten houden via hun overzichten. Want de elektra lijkt nu juist heel goedkoop op deze manier. 

Juist via de dsmrreader kun je dit inzicht wel zien!

dennissiemensma, 2022-02-10:
Bedankt voor je opheldering. Ik heb je dan eerder verkeerd begrepen! Je kunt dit dan wellicht het beste bij de leverancier aangeven wanneer je van mening bent dat ze daar iets aan kunnen verbeteren :rocket:

closed 2022-02-19 by dennissiemensma