dsmrreader/dsmr-reader #1581: 🙋Kan gene nieuwe virtualEnv aanmaken

D1ang, 2022-02-11:
### Description

Ik probeer naar v5 te upgraden echter, bij het uitvoeren van commando:
python3 -m venv ~/dsmr-reader/.venv/

Krijg ik de volgende error
`Error: Command '['/home/dsmr/dsmr-reader/.venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.`

Ik draai op een VM met ubuntu 18.04.5 met net naar python versie 3.9.9 geüpgraded
Enig idee hoe ik de Venv toch aangemaakt krijg?

### DSMR-reader version

4.20 to 5

### DSMR-reader platform

Native (e.g. manual installation)

D1ang, 2022-02-11:
Na wat verder spelen probleem kunnen oplossen.

Mocht iemand hetzelfde probleem krijgen; Ik miste Zlib.
na `sudo apt install zlib1g` en opnieuw de Python upgrade stappen uit te voeren ging het aanmaken van een vim wel.

closed 2022-02-11 by D1ang

dennissiemensma, 2022-02-12:
Bedankt voor je melding en de oplossing! Kun je je nog herinneren hoe je bij `zlib` uit kwam? Stond er nog een andere foutmelding bij?

D1ang, 2022-02-12:
Graag gedaan. 
Bij de Python 3.9.9 installatie op het punt: `sudo make install`
kreeg ik op het einde de volgende error [returned non-zero exit status 1]  toen in de console terug gescrolled en die gaf aan: 
`zlib not available`
daar op gezocht en kreeg ik wat oplossingen op Stackoverflow.