dsmrreader/dsmr-reader #1584: 🙋dsmr_consumption.services.run errored: index ...  contains unexpected zero page at block 1515

bennybubble, 2022-02-14:
### Description

[image]

Hi Dennis, since a few days (?) I see above error and it is not catching up.

Last few lines of the dsmr_backend;log file :

2022-02-14 09:51:08,507 ERROR    schedule     execute_scheduled_processes      37 | (InternalError) dsmr_consumption.services.run errored: index "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" contains unexpected zero page at block 1515
HINT:  Please REINDEX it.

2022-02-14 09:51:39,934 ERROR    schedule     execute_scheduled_processes      37 | (InternalError) dsmr_consumption.services.run errored: index "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" contains unexpected zero page at block 1515
HINT:  Please REINDEX it.

2022-02-14 09:52:11,489 ERROR    schedule     execute_scheduled_processes      37 | (InternalError) dsmr_consumption.services.run errored: index "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" contains unexpected zero page at block 1515
HINT:  Please REINDEX it.

2022-02-14 09:52:42,824 ERROR    schedule     execute_scheduled_processes      37 | (InternalError) dsmr_consumption.services.run errored: index "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" contains unexpected zero page at block 1515
HINT:  Please REINDEX it.

2022-02-14 09:53:14,191 ERROR    schedule     execute_scheduled_processes      37 | (InternalError) dsmr_consumption.services.run errored: index "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" contains unexpected zero page at block 1515
HINT:  Please REINDEX it.

Any idea on how to REINDEX this ?

### DSMR-reader version

4.20

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v4.20 / v3.9.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             410754
    Consumption records electricity / gas (est.)                                                56154 / 24662

UNRESOLVED ISSUES
    Too many unprocessed readings: 58900                                                  4 days, 2 hours ago
    Day statistics are lagging behind                                                    4 days, 10 hours ago
```

bennybubble, 2022-02-14:
Hoi Dennis,

Intussen verder gezocht en kwam een oude post tegen waarin een voorbeeld stond van het REINDEX commando, deze aangepast naar mijn 'error' en het lijkt te werken, even afwachten dus!

```
sudo su - postgres
psql dsmrreader
reindex index dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37;
```

In eerste instantie werkte eea en ander niet maar had de ";"  aan het einde over het hoofd gezien!

Wordt vervolgd!

dennissiemensma, 2022-02-14:
@bennybubble bedankt voor je vraag en update. Ik neem aan dat je #1448 bedoelt? 

Ik heb deze melding zelf nog niet meegemaakt en weet ook niet wat het veroorzaakt helaas. Je doet er wel goed om om een extra backup te maken en te bewaren, voor als er toch nog andere/grote issues naar voren komen.

Dit kan met:
```
sudo su - dsmr

./manage.py dsmr_backup_create --compact
./manage.py dsmr_backup_create --full
````

Vooral die tweede kan even duren, afhankelijk van je DB-grootte. Zorg er daarna voor dat je de backups wel van je storage afhaalt, als in dat je een kopie maakt of ze verplaatst, helemaal als je een SD-kaart gebruikt. De paden van de backups komen in de console te staan als het goed is.

bennybubble, 2022-02-15:
Bedankt voor de tip, alles werkt weer zoals het hoort.
De backups worden automatisch gemaakt naar een externe drive.

closed 2022-02-15 by bennybubble