dsmrreader/dsmr-reader #1585: Tweede slimme elektriciteitsmeter💡

jrosier, 2022-02-14:
### Feature

Ik heb 2 slimme meters: een gewone voor het verbruik en de teruglevering, en nog een brutoproductiemeter.
Is het mogelijk om een tweede slimme meter toe te voegen? Of moet ik dan een tweede (parallel) installatie uitvoeren?

dennissiemensma, 2022-02-14:
Bedankt voor je vraag. DSMR-reader heeft helaas geen ondersteuning voor meerdere meters. 

Wellicht dat dit ooit toegevoegd wordt, ook om een meterwissel wat makkelijker te maken, maar zeker niet op korte of middellange termijn. Dus dan blijft helaas alleen een tweede installatie voorlopig over als optie!

jrosier, 2022-02-14:
Ik begrijp het, er zullen weinig mensen zijn met 2 slimme meters...
Ik probeer nu een tweede installatie met user dsmr2 en database dsmr2, maar bij manage.py collectstatic --noinput probeert de installer te schrijven naar /var/www/dsmr (de eerste installatie) en geeft een foutmelding (geen schrijfrechten). Hoe kan ik dat verhelpen? Of moet ik gewoon beide installaties toegang geven tot /var/www/dsmr?

dennissiemensma, 2022-02-14:
Ik denk je dat ze het beste diezelfde map kunt laten delen. En ze telkens beiden update, zodat ze elkaar niet in de weg zitten.

dennissiemensma, 2022-02-14:
Overigens zijn er nog wel wat andere issues, want de instructies en configs gaan niet uit van een dubbele informatie op dezelfde host. Dus dat zal wel wat bijten. 

Je kunt nog wel overwegen om de Docker-variant van Xirixiz twee keer te draaien. Dan ben je niet of minder afhankelijk van botsingen op het OS, gezien de meeste logica in de Docker containers zit.

dennissiemensma, 2022-02-14:
Idem voor de Nginx config, de webserver, die luistert nu specifiek voor 1 installatie en stuurt dat ook door naar een hardcoded pad.

jrosier, 2022-02-15:
Zo te zien lukt het: drie supervisor-configuratiebestanden dubbel uitvoeren met andere namen; .env bestand aanpassen naar tweede database; tweede database inrichten; tweede gebruiker aanmaken; tweede nginx configuratiebestand aanmaken met andere poort en verwijzing naar andere socket. Voor de zekerheid een kopie gemaakt van /var/www/dsmrreader naar /var/www/dsmrreader2.

xirixiz, 2022-02-15:
Inderdaad, je kan het supervisor bestand dubbel uitvoeren en de naamegeving aanpassen. Nadeel is wel dat je daar rekening mee moet houden bij upgrades. Draai je al op v5.x? 

Een andere oplossing is door gebruik te maken van Docker. Dan kan je twee containers starten met ieder hun eigen functie zonder dat je supervisor bij hoeft te houden.

closed 2022-07-05 by dennissiemensma