dsmrreader/dsmr-reader #1590: 🙋 update from 4 to 5 is fails Step5

joopy, 2022-02-23:
### Description

Following the procedure on https://dsmr-reader.readthedocs.io/nl/v5/tutorial/upgrading/to-v5.html
Fails with the check any idea?

(.venv) dsmr@dsmr:~/dsmr-reader $ ./manage.py check
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    django.setup()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/apps/registry.py", line 122, in populate
    app_config.ready()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/contrib/admin/apps.py", line 27, in ready
    self.module.autodiscover()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/contrib/admin/__init__.py", line 24, in autodiscover
    autodiscover_modules('admin', register_to=site)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/module_loading.py", line 47, in autodiscover_modules
    import_module('%s.%s' % (app_config.name, module_to_search))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/dsmr_frontend/admin.py", line 7, in <module>
    from .models.settings import FrontendSettings, SortedGraph
  File "/home/dsmr/dsmr-reader/dsmr_frontend/models/settings.py", line 6, in <module>
    from colorfield.fields import ColorField
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/colorfield/fields.py", line 3, in <module>
    from colorfield.utils import get_image_file_background_color
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/colorfield/utils.py", line 3, in <module>
    from PIL import Image
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/PIL/Image.py", line 89, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

### DSMR-reader version

4.20

dennissiemensma, 2022-02-23:
Bedankt voor je melding. Had je deze ook al geprobeerd? Staat bovenaan bij hoofdstuk 5:

```
# You MAY NOT need this, but be sure to install it if you happen to hit the following error at any point later:
# "ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory"
sudo apt-get install libopenjp2-7-dev
```

Grote kans dat dat het oplost.

dennissiemensma, 2022-02-23:
Ik heb de docs wat verduidelijkt

joopy, 2022-02-23:
Ha Dennis,

Dank voor je snelle reply! Het zal idd de libopenjp2-7-dev geweest zijn. Ik kreeg geen foutmelding kan ik mij herinneren maar goed v5 draait nu hier. Vandaag zonnepanelen op het dak dus was er ff snel mee bezig. Dank voor het updaten van de documentatie deze kan dicht!

closed 2022-02-23 by joopy