dsmrreader/dsmr-reader #1591: 🙋DSMR = live consumption/returing values frozen

q144, 2022-02-27:
### Description

hi,
i recently clean reinstalled my pi2B with bullseye image and 5.0 release
all works fine except live consumption/return values are frozen. 
i do have historic data being filled in without delay.nothing wrong in logs, even with debug.
deploy works fine.no indication in support tab there is something wrong...
[image]

i think this info is generated via live-header-include.html but thats as far as i got.
any suggestions? 

### DSMR-reader version

5.0

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-02-27:
I suspect there is one future reading causing this, as dsmrreader uses the latest one for these data. Do you see any future readings in the admin in: /admin/dsmr_datalogger/dsmrreading/

q144, 2022-02-27:
no,

most recent are from today

dennissiemensma, 2022-02-27:
What does this output?

```
sudo su - postgres
psql dsmrreader

SELECT id, timestamp, electricity_delivered_1 FROM dsmr_datalogger_dsmrreading ORDER BY timestamp DESC LIMIT 10;
```

q144, 2022-02-28:
id    |       timestamp        | electricity_delivered_1
----------+------------------------+-------------------------
  49115537 | 2022-02-28 09:03:53+01 |                3888.367
  49115536 | 2022-02-28 09:03:48+01 |                3888.364
  49115535 | 2022-02-28 09:03:42+01 |                3888.361
  49115534 | 2022-02-28 09:03:37+01 |                3888.358
  49115533 | 2022-02-28 09:03:31+01 |                3888.355
  49115532 | 2022-02-28 09:03:26+01 |                3888.352
  49115531 | 2022-02-28 09:03:20+01 |                3888.349
  49115530 | 2022-02-28 09:03:15+01 |                3888.346
  49115529 | 2022-02-28 09:03:09+01 |                3888.342
  49115528 | 2022-02-28 09:03:03+01 |                3888.339

digiater, 2022-02-28:
I've got exactly the same issue in v5.0:

dsmrreader=# SELECT id, timestamp, electricity_currently_delivered, electricity_currently_returned FROM dsmr_datalogger_dsmrreading ORDER BY timestamp DESC LIMIT 10;
   id    |       timestamp        | electricity_currently_delivered | electricity_currently_returned
---------+------------------------+---------------------------------+--------------------------------
 4394568 | 2022-02-28 15:07:02+01 |                           0.000 |                          2.245
 4394567 | 2022-02-28 15:06:47+01 |                           0.000 |                          2.176
 4394566 | 2022-02-28 15:06:35+01 |                           0.000 |                          2.225
 4394565 | 2022-02-28 15:06:22+01 |                           0.000 |                          2.161
 4394564 | 2022-02-28 15:06:10+01 |                           0.000 |                          2.221
 4394563 | 2022-02-28 15:05:54+01 |                           0.000 |                          2.321
 4394562 | 2022-02-28 15:05:42+01 |                           0.000 |                          2.218
 4394561 | 2022-02-28 15:05:31+01 |                           0.000 |                          2.168
 4394560 | 2022-02-28 15:05:20+01 |                           0.000 |                          2.269
 4394559 | 2022-02-28 15:05:07+01 |                           0.000 |                          2.252
(10 rows)

This is an old installation recently upgraded to v5.0. I just noticed this today as my consumption is frozen on 2070 and returning on 1111. The graphs seem to be okay.

q144, 2022-02-28:
i did try to fool/force the live update by upgrading a record and restart services but no change 
(eg:update dsmr_datalogger_dsmrreading set timestamp='2-2-2023 18:01:14+01' where id =49121125;)

dennissiemensma, 2022-02-28:
@q144 what does the updated query return?

```
SELECT id, timestamp, electricity_currently_delivered, electricity_currently_returned FROM dsmr_datalogger_dsmrreading ORDER BY timestamp DESC LIMIT 10;
```

q144, 2022-02-28:
id    |       timestamp        | electricity_currently_delivered | 
electricity_currently_returned
----------+------------------------+---------------------------------+--------------------------------
  49121127 | 2023-02-02 18:01:14+01 | 0.388 |                          0.000
  49121645 | 2022-02-28 18:49:31+01 | 0.490 |                          0.000
  49121644 | 2022-02-28 18:49:26+01 | 0.490 |                          0.000
  49121643 | 2022-02-28 18:49:20+01 | 0.494 |                          0.000
  49121642 | 2022-02-28 18:49:15+01 | 0.487 |                          0.000
  49121641 | 2022-02-28 18:49:09+01 | 0.499 |                          0.000
  49121640 | 2022-02-28 18:49:04+01 | 0.509 |                          0.000
  49121639 | 2022-02-28 18:48:58+01 | 0.500 |                          0.000
  49121638 | 2022-02-28 18:48:53+01 | 0.497 |                          0.000
  49121637 | 2022-02-28 18:48:47+01 | 0.497 |                          0.000

dennissiemensma, 2022-02-28:
@q144 @digiater can you press F12 while viewing the page and 

- check whether the `header` calls succeed in the **network** tab?
- and if so, view the response of them?

[image]

----

[image]

q144, 2022-02-28:
found it.
i disabled and re enabled tracking protection and now its fine.
i believe the live data was marked as "cached".
this action probably reset/deleted my cache for this site

thanks :)

/feels like looking for a needle in a haystack while you're sitting on it...

Message ID: ***@***.***>
{
  "log": {
    "version": "1.2",
    "creator": {
      "name": "Firefox",
      "version": "97.0.1"
    },
    "browser": {
      "name": "Firefox",
      "version": "97.0.1"
    },
    "pages": [
      {
        "startedDateTime": "2022-02-28T19:16:03.100+01:00",
        "id": "page_3",
        "title": "Dashboard | DSMR Reader",
        "pageTimings": {
          "onContentLoad": -1,
          "onLoad": -1
        }
      }
    ],
    "entries": [
      {
        "pageref": "page_3",
        "startedDateTime": "2022-02-28T19:16:03.100+01:00",
        "request": {
          "bodySize": 0,
          "method": "GET",
          "url": "http://10.0.1.5:81/xhr/header",
          "httpVersion": "HTTP/1.1",
          "headers": [
            {
              "name": "Host",
              "value": "10.0.1.5:81"
            },
            {
              "name": "User-Agent",
              "value": "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:97.0) Gecko/20100101 Firefox/97.0"
            },
            {
              "name": "Accept",
              "value": "application/json, text/javascript, */*; q=0.01"
            },
            {
              "name": "Accept-Language",
              "value": "en-US,en;q=0.5"
            },
            {
              "name": "Accept-Encoding",
              "value": "gzip, deflate"
            },
            {
              "name": "Referer",
              "value": "http://10.0.1.5:81/"
            },
            {
              "name": "X-Requested-With",
              "value": "XMLHttpRequest"
            },
[... 197 lines cut ...]
          "content": {
            "mimeType": "application/json",
            "size": 119,
            "text": "{\"timestamp\": \"now\", \"currently_delivered\": 550, \"currently_returned\": 0, \"cost_per_hour\": null, \"tariff_name\": \"High\"}"
          },
          "redirectURL": "",
          "headersSize": 0,
          "bodySize": 119
        },
        "cache": {},
        "timings": {
          "blocked": 0,
          "dns": 0,
          "ssl": 0,
          "connect": 0,
          "send": 0,
          "wait": 0,
          "receive": 0
        },
        "time": 0,
        "_securityState": "insecure"
      }
    ]
  }
}

dennissiemensma, 2022-02-28:
Thanks for sharing! I'll add some no-caching header in the next release. Maybe that will solve it by default.

dennissiemensma, 2022-02-28:
Current release:

[image]

----

Next release:

[image]

---

Hopefully it will prevent browser caching or whatever.

closed 2022-02-28 by dennissiemensma

digiater, 2022-02-28:
[image]

The response is indeed cached. But I do not seem te get rid of it, restarting the browser, disabling/enabling enhanced protection, clearing the browser cache, restarting the webserver on the dsmr system all do not help.

dennissiemensma, 2022-02-28:
DSMR-reader v5.1 is scheduled for tomorrow anyway, so it may be fixed shortly

dennissiemensma, 2022-03-01:
Released in https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.1.0

dennissiemensma, 2022-03-02:
Can you check in the network tab whether the Expires/Cache-Control response headers are sent? 
And are there any other errors on the console tab? 

It works fine for me in Firefox 97 on Ubuntu, but it already did on previous DSMR-reader versions as well. So it quite hard for me to replicate the issue.

[image]

digiater, 2022-03-02:
It's not there. Notice the date in the response header, it's old. (the clock on the system has the correct time).

[image]

digiater, 2022-03-02:
There's a little button in the debug pane that says 'disable cache' and that did the trick. After that the entries will be fetched again from the server. This appears to be a browser issue.

[image]

dennissiemensma, 2022-03-02:
Great, hopefully it'll fix the problem forever