dsmrreader/dsmr-reader #1592: 🙋Restore backup in a docker DSMR

Rob99999, 2022-03-02:
### Description

Ik heb een docker container draaien met DSMR (in proxmox). Door een onzorgvuldige update (DSMR 4 naar 5 en changelog niet goed gelezen ) was de database niet meer juist. En vervolgens van de ene loophole in de andere. Uiteindelijk een geheel nieuwe install gerealiseerd die nu werkt. Nu ben ik mijn data kwijt. Gelukkig heb ik een back-up via Dropbox veiliggesteld. 

Na veel lezen ben ik nog onzeker hoe ik deze restore kan uitvoeren. Ik ben niet heel goed thuis in een linux omgeving. Heb wel  een bericht gelezen van [xirixiz](https://github.com/xirixiz) commented [on 1 Oct 2018](https://github.com/dsmrreader/dsmr-reader/issues/490#issuecomment-425857544) • waar volgens mij de juiste route is weergegeven.

Mijn SQL heeft echter een Gz. extensie. Moet ik deze eerst uitpakken. Moet ik de naam van de Sql db aanpassen naar het commando. Hoe krijg ik deze vervolgens in de juiste folder (in proxmox) in docker. en klopt het dat ik dan alleen nog de weergegeven commando's hoef uit te voeren?

### DSMR-reader version

DSMR-reader v5.1

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-03-02:
Bedankt voor je vraag. Je kunt dit het beste voorleggen bij of aan @xirixiz want ik draai zelf (nog) geen Docker en heb ook geen restore hoeven uit te voeren daarmee.

xirixiz, 2022-03-03:
Hoi Rob, je kan de tgz idd uitpakken en dan de restore instructies volgen: 
```tar xvzf your_file.tar.gz```

Rob99999, 2022-03-03:
Bedankt voor jullie antwoorden. Via WinSCP naar Proxmox en met PCT push sql bestand naar DSMR_backups folder geplaatst. @xirixiz ; het is geen tar.gz maar en sql.gz als ik het goed zie. Ik heb het bestand derhalve via 7zip in windows uitgepakt en vervolgens verplaatst. nu loop ik echter tegen het volgende issue aan:
Container gestopt en vervolgens 2de commando ingegeven
root@Docker ~/Compose# docker exec -t dsmrdb dropdb dsmrreader -U dsmrreader
Error: No such container: dsmrdb
root@Docker ~/Compose# dir
docker-compose.yml  dsmr_backups  dsmrdb  unifi
root@Docker ~/Compose# cd dsmrdb
root@Docker ~/Compose/dsmrdb# docker exec -t dsmrdb dropdb dsmrreader -U dsmrreader
Error: No such container: dsmrdb  

Er zou geen dsmrdb container zijn. ik begrijp de foutmelding niet.

xirixiz, 2022-03-03:
Ah, ok, dan is het een gzip bestand, die kan je idd met gunzip of 7zip of een andere willekeurige tool uitpakken.
Welke Docker containers heb je draaien voor dsmr? In de voorbeelden die ik gebruik wordt er standaard een dsmrdb container aangemaakt, maar bij jou heeft die container een andere naam. Je moet de naam van de DSMR Docker database container achterhalen en dat gebruiken ipv dsmrdb.

Met ```docker ps -a``` kan je wel achterhalen welk id of naam de container heeft.

Rob99999, 2022-03-04:
Ik heb wel de folder en kan het bestand erna uploaden. als ik dan DSMR stop en vervolgens het commando " docker exec -t dsmrdb createdb -O dsmrreader dsmrreader -U dsmrreader" invoer krijg ik een foutmelding dat de dsmrdb container aan het opstarten is. In de log krijg ik deze foutmelding:  

PostgreSQL Database directory appears to contain a database; Skipping initialization

2022-03-05 00:03:32.731 CET [1] FATAL:  database files are incompatible with server
2022-03-05 00:03:32.731 CET [1] DETAIL:  The data directory was initialized by PostgreSQL version 12, which is not compatible with this version 13.6.