dsmrreader/dsmr-reader #1597: Geen Gas informatie in de DSMR, wel in de logger🚨

Milithor, 2022-03-12:
### Description

Ik heb een nieuwe server geïnstalleerd en overgegaan op een clean install van DSMRv5.
Hierbij valt op dat er geen gast in de tool wordt opgeslagen, voorheen werkte dit wel.

Ik heb een remote datalogger, ook geüpdatet naar v5.

```
0-0:96.1.4(50216)
0-0:96.1.1(3153414733313030303530313836)
0-0:1.0.0(220312120532W)
1-0:1.8.1(003727.851*kWh)
1-0:1.8.2(004196.698*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.376*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.477*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.060*kW)
1-0:41.7.0(00.382*kW)
1-0:61.7.0(00.033*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:32.7.0(240.1*V)
1-0:52.7.0(238.6*V)
1-0:72.7.0(236.1*V)
1-0:31.7.0(000.26*A)
1-0:51.7.0(001.90*A)
1-0:71.7.0(000.39*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313139303830313935)
0-1:24.4.0(1)
0-1:24.2.3(220312120501W)(01726.969*m3)
```

De laatste log lijn die binnenkomt op de DSMR vertelt nochtans dat er een gasmeting is (0126.969*m3).

Doe ik wat verkeerd? 'Hide Gas' in de backend opties staat uit.

### DSMR-reader version

5.1

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-03-12:
Bedankt voor je melding. Wat is de output van de debug info dump? https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/debug-info-dump.html

En kan het zijn dat je een niet-NL meter hebt? Als dat zo is dan zul je die ook nog moeten instellen bij de datalogger settings.

Milithor, 2022-03-12:
*Slaps his own face, bows and apologizes*
Ik heb een remote datalogger, dus heb niet direct naar die datalogger config gekeken.
Na herstart van DSMR krijg ik nu wel gasmetingen, waarvoor dank!

closed 2022-03-12 by Milithor