dsmrreader/dsmr-reader #1600: 🚨Can only connect with DSMR-5S

spafruit1, 2022-03-18:
### Description

My P1 reader was working fine with DSMR 5 untill it stopped working a week. Now i can only connect with DSMR 5S but i get no data i got only 0 value's :( The P1 reader is stil reachable with telnet. 

Any suggetions on what i can try next?

### DSMR-reader version

5 and 5S

dennissiemensma, 2022-03-19:
Hi I'm not sure what 5s is. Did you try using the cu program as well? If it suddenly broke down and cu does not work either, there is a high possibility that the reader or cable stopped working.

spafruit1, 2022-03-19:
5s means DSMR version 5s (s means Swedish but mine is dutch). What is the cu program? I am able to have communication via CMD telne
[image]
[image]
t

dennissiemensma, 2022-03-20:
The cu program can be used to read serial cables. If you're not using a cable but some other appliance, it should be fine without. Using telnet or cu allows you to check what dsmrreader reads. It seems your telegram has some corruption with weird symbols. I'm not sure if 5s uses other symbols, but dsmrreader does not support these for sure. You may want to check for any signal interference causing these issues.

spafruit1, 2022-03-20:
I got this weird symbols in both 5 and 5s. Maybe that is my problem i go try a new cable and let you know