dsmrreader/dsmr-reader #1602: Graph numbers not showing in light theme

leadenchopsticks, 2022-03-21:
### Description

After upgrading to 5.1 the GUI reverted to dark (probably due to my Mac settings). After manually changing to light theme, the numbers on the left and bottom of the graphs are not visible anymore.

### DSMR-reader version

5.1

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-03-21:
Would you be able to attach a screenshot of the issue?

leadenchopsticks, 2022-03-22:
Sure!

Working:

[image]

Not working:

[image]

dennissiemensma, 2022-03-22:
Unfortunately I'm unable to reproduce the issue in both Firefox or Chrome. 

Which browser are you using? Are you able to try another browser? Are there any errors on the Developer Tools' Console tab of your browser?

leadenchopsticks, 2022-03-23:
It happens in both Chrome and Safari. No errors in developer consoles. My default Mac setting is 'Dark'. 

Interestingly enough, when I change my Mac setting to 'Light' the numbers are showing up. So it seems to be an issue when manually selecting Light theme in DSMR reader in combination with Dark mode in OS.

dennissiemensma, 2022-03-23:
Great, I can now see the issue here as well. I've fixed the issue for v5.2

dennissiemensma, 2022-03-23:
Fixed in 900f8ab023475d34223e85473cd74f58ad3c0882 and will default to light mode in the future on error as well.

closed 2022-03-23 by dennissiemensma

dennissiemensma, 2022-05-24:
Net gereleased via [v5.2](https://dsmr-reader.readthedocs.io/en/v5/reference/changelog.html#v5-2-0-2022-05-24)