dsmrreader/dsmr-reader #1604: 🙋Error during migration from v4 to v5 ./manage.py migrate (No Password Supplied)

wesleykappen, 2022-03-23:
### Description

Hi,

During the migration from v4 to v5 I've ran into the following issue: 

executing: `(.venv) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py migrate` results in the following error:

`Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: fe_sendauth: no password supplied

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/base.py", line 89, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 75, in handle
    self.check(databases=[database])
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/management/base.py", line 423, in check
    databases=databases,
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/checks/registry.py", line 76, in run_checks
    new_errors = check(app_configs=app_configs, databases=databases)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/core/checks/model_checks.py", line 34, in check_all_models
    errors.extend(model.check(**kwargs))
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/models/base.py", line 1303, in check
    *cls._check_indexes(databases),
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/models/base.py", line 1695, in _check_indexes
    connection.features.supports_covering_indexes or
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/postgresql/features.py", line 93, in is_postgresql_11
    return self.connection.pg_version >= 110000
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 329, in pg_version
    with self.temporary_connection():
  File "/usr/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 603, in temporary_connection
    with self.cursor() as cursor:
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.7/site-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: fe_sendauth: no password supplied
`

### DSMR-reader version

4.20 (latest v4) version

wesleykappen, 2022-03-23:
Hi, 

Just after posting this, i've found the issue. It was a typo in .env file. I had DJANGO_DATEBASE_PASS instead of DJANGO_DATABASE_PASSWORD.

Sorry! 

Regards,

Wesley.

dennissiemensma, 2022-03-23:
Fijn dat je het zelf gevonden hebt!

closed 2022-03-23 by dennissiemensma