dsmrreader/dsmr-reader #1607: 💡 Feature request: variabele energiecontracten van bijv ANWB

Niek, 2022-03-24:
### Feature

Er zijn een aantal energieaanbieders die "real-time" variabele tarieven rekenen, zoals bijv ANWB en Frank Energie. De tarieven variëren per uur en zijn via een API (meestal ongedocumenteerd) op te vragen - een dag van tevoren worden de nieuwe tarieven bekend. Hieronder 2 voorbeelden:

ANWB:
```shell
curl -sf 'https://api.energyzero.nl/v1/energyprices?fromDate=2022-01-01T00%3A00%3A00.000Z&tillDate=2022-12-31T23%3A59%3A59.999Z&interval=4&usageType=1&inclBtw=true'
```
Frank Energie:
```shell
curl -sf 'https://frank-graphql-prod.graphcdn.app/' -d '{"query":"query MarketPrices($startDate: Date!, $endDate: Date!) { marketPricesElectricity(startDate: $startDate, endDate: $endDate) { till from marketPrice marketPriceTax sourcingMarkupPrice } marketPricesGas(startDate: $startDate, endDate: $endDate) { from till marketPrice marketPriceTax sourcingMarkupPrice } }","variables":{"startDate":"2022-03-01","endDate":"2022-03-31"},"operationName":"MarketPrices"}' -H 'Content-Type: application/json'
```

Is het een idee support voor een aantal van deze energieaanbieders te integreren? Of als alternatief, een API methode te implementeren om de energiecontracten te updaten?

dennissiemensma, 2022-03-24:
Bedankt voor je verzoek. Ik zou de API er ooit voor kunnen uitbreiden, echter rekent DSMR-reader voor de dagtotalen slechts 1x per dag uit wat de kosten zijn, dus gaat dat sowieso niet werken voor verschillende tarieven per uur.

Verder is het doel ook nooit geweest om de tarieven helemaal te spiegelen, hooguit een grove indicatie wat "een dag je ongeveer kost", tov bijvoorbeeld wel/geen verwarming aan etc.

Niek, 2022-03-24:
OK, dat maakt het inderdaad wel lastig. Wellicht is het dan handiger de calculatie in bijv PVOutput te doen, hoewel die ook nog geen uurlijke prijzen ondersteunen: https://forum.pvoutput.org/t/api-enhancement-to-change-tariff/5047

closed 2022-03-26 by dennissiemensma