dsmrreader/dsmr-reader #1609: 💡 Meerdere DSMR instantie backuppen naar dropbox

n9iels, 2022-04-01:
### Feature

Voorheen was het mogelijk om 2 DSMR-reader instantie te laten back-uppen naar een dropbox account. Als ik het goed heb kan dit nu niet meer omdat het allemaal in dezelfde map "Official DSMR-Reader" terrecht komt i.p.v. een map per instantie. Klopt dit? 🤔

dennissiemensma, 2022-04-01:
Bedankt voor je vraag. Als het goed is was het voorheen ook niet mogelijk om dit te doen, tenzij je expliciet twee verschillende "Dropbox Apps" gebruikte per installatie. Het was dan ook meer een bijeffect dan een feature.

Ik heb in v5.x de Dropbox flow ingebouwd zoals zij dat van developers verwachten. Het was niet de bedoeling dat iedereen een eigen Dropbox App maakte voor hun DSMR-reader-installatie en technisch gezien ook niet nodig.
De backups in DSMR-reader houden echter geen rekening met meerdere installaties, dus voor nu is het denk ik het beste om simpelweg bij 1 van je 2 installaties de Dropbox App key in DSMR-reader te overschrijven.

Je kunt dit doen door in je `.env` of als envvar (als je via Docker draait) je eigen Dropbox App Key in `DSMRREADER_DROPBOX_APP_KEY` te definieren:
- https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmrreader/config/defaults.py#L54

Dit is bewust ongedocumenteerde functionaliteit, dus meer als nood nu. Ik zal voor een volgende release kijken of ik een setting kan maken zodat je een prefix kan instellen voor de naam van de backup-bestanden.

Vergeet niet om DSMR-reader te herstarten en zorg ook dat je in Dropbox app zelf `Allow public clients (Implicit Grant & PKCE)` aan hebt staan:

[image]

dennissiemensma, 2022-04-01:
Vanaf v5.2 kun je `DSMRREADER_BACKUP_NAME_PREFIX` gebruiken om de prefix van de backupbestanden aan te passen.  Standaardwaarde is `dsmrreader` uit onderstaande voorbeelden.

Bijvoorbeeld:
```
dsmrreader-postgresql-partial-backup-2022-04-01.sql
```

Wordt dan:
```
dsmrreader01-postgresql-partial-backup-2022-04-01.sql
dsmrreader02-postgresql-partial-backup-2022-04-01.sql
```

Met:
```
DSMRREADER_BACKUP_NAME_PREFIX=dsmrreader01
DSMRREADER_BACKUP_NAME_PREFIX=dsmrreader02
```

Dan heb je bovenstaande workaround ook niet meer nodig.

closed 2022-04-01 by dennissiemensma

n9iels, 2022-04-02:
Hey Dennis, wat een enthousiasme zeg! Zodra v5.2 uitkomt ga ik dit proberen, thanks en een heel fijn weekend!