dsmrreader/dsmr-reader #1612: The headers or library files could not be found for jpeg

RichieB2B, 2022-04-06:
### Description

Tijdens de upgrade naar v5.1.0 kon bij de stap
```
pip3 install -r ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
```
Pillow niet worden geïnstalleerd door deze foutmelding:
```
      The headers or library files could not be found for jpeg,
      a required dependency when compiling Pillow from source.
```
De oplossing was simpel:
```
apt-get install libjpeg-dev
```

### DSMR-reader version

5.1.0

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-04-06:
Bedankt voor je melding. Werkte de `sudo apt-get install libopenjp2-7-dev` lib uit de [upgrade guide](https://dsmr-reader.readthedocs.io/en/v5/tutorial/upgrading/to-v5.html#upgrade-to-dsmr-reader-v5) niet?

Zo ja, welk OS met welke OS-versie gebruik je?

RichieB2B, 2022-04-06:
Jawel, die deed het prima. Ik draai raspbian buster op Pi 3b.

dennissiemensma, 2022-04-07:
Bedankt voor je aanvulling. Wellicht begrijp ik je verkeerd hoor, maar werkte alleen `libjpeg-dev` voor je en niet `libopenjp2-7-dev`? Of is de ene een alias van de andere?

RichieB2B, 2022-04-07:
Ik heb `libopenjp2-7-dev` geïnstalleerd zoals in de instructies staat. Pillow heeft echter (ook?) `libjpeg-dev` nodig.

dennissiemensma, 2022-04-07:
Hmm vreemd. Wat krijg je allemaal te zien bij deze:

```
sudo apt list --installed | grep jp
```

Ik heb ook Buster en blijkbaar werkend via deze:
```
libjpeg62-turbo/oldstable,now 1:1.5.2-2+deb10u1 armhf [installed,automatic]
libopenjp2-7-dev/oldstable,now 2.3.0-2+deb10u2 armhf [installed]
libopenjp2-7/oldstable,now 2.3.0-2+deb10u2 armhf [installed,automatic]
```

Staat er bij jou wat anders bij?

dennissiemensma, 2022-04-07:
Ik heb overigens tot nu toe drie meldingen gehad, dus ik weet alleen wat er gemeld wordt en niet wat men zelf oplost.

-  `libopenjp2-7-dev` - #1590
-  `libopenjp2-7-dev` - #1577
- `libjpeg-dev` - jouw melding

RichieB2B, 2022-04-07:
`ImportError: libopenjp2.so.7` lijkt me ongerelateerd aan `The headers or library files could not be found for jpeg` ?

dennissiemensma, 2022-04-07:
Ja de melding wel, alleen kwamen die andere fouten ook uit Pillow en ze hebben beiden te maken met JPEG. Dus ik ben zoekende waarom het bij mij wel werkte of installeerde zonder die extra lib. Op hetzelfde OS.

dennissiemensma, 2022-04-07:
Welke Python-versie draai je? Wellicht zit daar nog een (conditioneel) verschil in voor Pillow. Ik zit op Python 3.9.

RichieB2B, 2022-04-07:
Buster heeft Python 3.7.3. Probeer dit eens:
```
docker run -it debian:buster /bin/bash
apt update
apt -y install git python3-dev python3-venv gcc libpq-dev zlib1g-dev
git clone --branch v5 https://github.com/dsmrreader/dsmr-reader.git
python3 -m venv ~/dsmr-reader/.venv/
source ~/dsmr-reader/.venv/bin/activate
cd dsmr-reader/
pip3 install -r dsmrreader/provisioning/requirements/base.txt
```
Dit geeft:
```
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-kiheff18/Pillow/setup.py", line 1009, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
```
Het package `libjpeg-dev` is gewoon nodig voor Pillow op buster. Ik krijg trouwens geen foutmelding voor een missende `libopenjp2.so.7`.

RichieB2B, 2022-04-07:
> Welke Python-versie draai je? Wellicht zit daar nog een (conditioneel) verschil in voor Pillow. Ik zit op Python 3.9.

Installeert met Python 3.9 het Pillow package wellicht als `whl` of wordt hij ook van source gecompileerd? Want alleen dan heb je `libjpeg-dev` nodig natuurlijk.

dennissiemensma, 2022-04-08:
Bedankt voor je aanvullingen! Ik heb hier nog wat dieper naar gekeken en ik kom er niet uit. 

Via mijn docker dev setup op Buster kan ik het niet reproduceren, maar wel met jouw stappen in Buster. Met dezelfde stappen werkt het wel weer zonder die fout op Bullseye.
Het kan best zijn dat het in Python 3.9 anders werkt of dat Pillow daar anders wordt geinstalleerd en daarom ook geen issue is bij mijn installatie en wellicht ook anderen die bijvoorbeeld al Debian Bullseye (of gewoon Docker) gebruiken. Overigens hint de upgrade-guide ook al op Python 3.9, dus wellicht nog een andere reden dat er niemand nog tegenaan is gelopen of gemeld heeft.

Ik zal voor de volgende release een heads up plaatsen, mocht iemand er ook tegenaan lopen bij Python 3.7.

closed 2022-04-08 by dennissiemensma