dsmrreader/dsmr-reader #1613: 🙋 Energy used during power outage

joopy, 2022-04-07:
### Description

Today there was a power outage in the street from 9:08 till 10:11 during that time there is registered power usage? The dsmr raspberry pi was also off (no power) when I look in the graph I do see power usage?
[http://photos.noraliedevries.nl/upload/2022/04/07/20220407091924-f1734cc5.jpg](url)

### DSMR-reader version

5.0

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.0 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             148666
    Consumption records electricity / gas (est.)                                                98249 / 31821

UNRESOLVED ISSUES
    Database groeit gestaag: 3250 MB, overweeg dataopschoning (indien nog niet ingeschakeld)                                      nu

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                2388 MB
    public.dsmr_consumption_electricityconsumption                                                     836 MB
```

niek1992, 2022-04-07:
Looks like Power Returned to the grid? So you have solar panels that might communicate with your DSMR after power was restored? might be incorrect though, am not familiar with solar panels and returning it to the grid when there's an outage.

dennissiemensma, 2022-04-07:
Bedankt voor je vraag. Als je goed naar de tijden kijkt zie je ook de overgang bij de storing, alleen laat DSMR-reader geen gaten zien in de grafiek.

[image]

De schaal is ongeveer hier:
- 5 minuten
- 1 uur en 2 minuten (de storing)
- 5 minuten

Dit is een bekend issue in #1278, maar ik heb er geen makkelijke oplossing voor qua visualisatie.

closed 2022-04-07 by dennissiemensma

joopy, 2022-04-07:
Dank voor je uitleg! Ik dacht even dat ik gek werd.