dsmrreader/dsmr-reader #1616: 💡Uitgebreider archief

nwametze, 2022-04-21:
### Feature

Ik zie dat er steeds meer belangstelling komt om analyses te maken in relatie tot stroomopslag en peakshaving. Hiervoor mis ik gedetailleerde informatie in het archief.  
Informatie die ik momenteel daarvoor mis in het archief:
- max amperage verbruik per fase per dag (peakshaving);
- totaal kWh verbruik per fase per dag (opslag);
- totaal kWh teruglevering per fase per dag (opslag). 
Is er wellicht al over nagedacht om dit soort gedetailleerde informatie op verzoek op te laten slaan en inzichtelijk te maken? Levert natuurlijk meer data op die opgeslagen moet worden, maar als het optioneel gemaakt kan worden denk ik dat het een meerwaarde kan zijn voor het project.

dennissiemensma, 2022-04-21:
Bedankt voor je verzoek. Het hangt er vooral vanaf welke gegevens er beschikbaar zijn. De dataopslag is niet zo'n issue, daar kan iedereen zelf een keuze in maken qua resolutie met ofwel de datalogger-sleep aan de "invoer" zijde of retentie/opschoning achteraf.

Tenzij er recente ontwikkelingen in zijn, dacht ik dat de gegevens die je vraagt niet allemaal beschikbaar zijn in de slimme meter. Tenminste, niet **per fase**.

Als het goed is is alleen onderstaande per fase beschikbaar, zoals Ampere, Volt en Wattages, maar niet kWh.

```
1-0:32.7.0(220.1*V)
1-0:52.7.0(220.2*V)
1-0:72.7.0(220.3*V)

1-0:31.7.0(001*A)
1-0:51.7.0(002*A)
1-0:71.7.0(003*A)

1-0:21.7.0(01.111*kW)
1-0:41.7.0(02.222*kW)
1-0:61.7.0(03.333*kW)

1-0:22.7.0(04.444*kW)
1-0:42.7.0(05.555*kW)
1-0:62.7.0(06.666*kW)
```

Via #1084 komt er trouwens wel een afgeleide bij straks, het kwartiergemiddelde. Het is iets wat voor Belgische gebruikers relevant is per 1 juli, maar in zeker zin kan iedereen daar wel piekverbruik aan afleiden voor een grove indicatie.

Alleen voor de Amperes is dus uiteindelijk wel iets te maken of bij te houden vermoed ik. Al heb je wellicht ook mogelijk indirect wat aan bovenstaande.

nwametze, 2022-05-06:
Dank voor je reactie. Ik zal #1084 in de gaten houden. Maar top dat je deze in ieder geval nog even op de toekomst gezet hebt.

dennissiemensma, 2022-07-06:
Ik heb hier nog opnieuw naar gekeken:
- **max amperage verbruik per fase per dag (peakshaving);**

Uiteindelijk lijkt het me wat breder inzetbaar als je hiervoor de API of MQTT gebruikt. De ene gebruiker zal het namelijk per dag willen, de andere per uur of week, etc.

In principe is het (vooral met MQTT) relatief makkelijk om de hoogste waarde van de huidige X periode (in dit geval "dag") bij te houden. Je kunt dan zelfs er een alert of iets op zetten wanneer een van de fases over een threshold heen gaat.

Wellicht dat dit in HA of een andere tool die op MQTT topics kan luisteren.

---

Dit is vergelijkbaar met #1655. Daar geef ik ook nog een derde optie, namelijk door het (achteraf) in de database op te vragen. daar ging het over spanning, maar in jouw geval wordt het zoiets voor Amperes:

```
sudo su - postgres
psql dsmrreader

select date_trunc('day', timestamp) "day", MAX(phase_power_current_l1) as max_l1, MAX(phase_power_current_l2) AS max_l2, MAX(phase_power_current_l3) AS max_l3 from dsmr_datalogger_dsmrreading group by day order by day desc limit 50;

# Of als losse regels

select date_trunc('day', timestamp) "day", MAX(phase_power_current_l1) as max_l1, MAX(phase_power_current_l2) AS max_l2, MAX(phase_power_current_l3) AS max_l3 
from dsmr_datalogger_dsmrreading 
group by day 
order by day desc 
limit 50;
```

Dus dezelfde query, maar dan voor `phase_power_current_l1` / `phase_power_current_l2` / `phase_power_current_l3`. Zie ook https://github.com/dsmrreader/dsmr-reader/issues/1655#issuecomment-1175394903

Daarnaast kun je ook relatief eenvoudig andere

closed 2022-07-06 by dennissiemensma