dsmrreader/dsmr-reader #1619: 🙋 ERROR:  null value in column "delivered_1"

randommen96, 2022-04-23:
### Description

In een van de twee installaties die ik beheer loop ik tegen het volgende probleem aan:
```
**Too many unprocessed readings: 318** (**47 minutes ago**)

**Day statistics are lagging behind** (**2 weeks, 5 days ago**)
```
Het volgende issue heb ik al gevonden en geprobeerd: https://github.com/dsmrreader/dsmr-reader/issues/1288

Vervolgens wordt er 1 row verwijderd.

Als ik daarna dsmrreader weer start, geeft de runner die de statistics aanmaakt problemen, die denkt dat er nog steeds NULL waardes aanwezig zijn, dit terwijl een DELETE query 0 rows geeft en SELECT ook 0 rows...

Wel valt op dat er continue in de logging voorkomt dat er NULL waardes geprobeerd worden te inserten, wat niet mag, de TimescaleDB logs bevestigen dit:
```
2022-04-20 20:29:03.881 UTC [709] STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2022-04-01T19:24:00+00:00'::timestamptz, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"
2022-04-20 20:29:34.680 UTC [709] ERROR:  null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
```

Is er nog iets anders dat ik kan proberen? Ik heb inmiddels de stappen een aantal keer doorlopen maar ben ten einde raad, alle instellingen heb ik vergeleken met de wel werkende installatie, het enige verschil is dat de één geen teruglevering doet en de probleeminstallatie wél.

Ik draai het binnen homeassistant met de addon van sanderdw + USB P1 kabel.

De live grafieken lijken overigens prima te werken, ook is de data die binnenkomt zeker geen 0 waarde.

De tijdzones heb ik ook vergeleken, hierin is interessant dat de telegram tijd 2 uur achterloopt t.o.v. normale Nederlandse tijd, dit krijg ik alleen niet rechtgetrokken.

Gr,
David :-)

### DSMR-reader version

5.1

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.1 / v3.10.4 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             307908
    Consumption records electricity / gas (est.)                                                47137 / 24012

UNRESOLVED ISSUES
    Too many unprocessed readings: 1339                                                    3 weeks, 1 day ago
    Day statistics are lagging behind                                                      3 weeks, 1 day ago
```

dennissiemensma, 2022-04-24:
Bedankt voor je melding. Hoe krijg je de metingen binnen, direct via de kabel die DSMR-reader dan zelf uitleest?

Je zou ook nog kunnen kijken in de tabel met metingen, of daar NULL-waardes staan:
```
SELECT * FROM dsmr_datalogger_dsmrreading WHERE electricity_delivered_1 IS NULL;
```

randommen96, 2022-04-24:
Klopt ja, direct via USB naar de DMSR-reader container.
Heb even gekeken en die geeft ook 0 rows terug, nog andere ideeen?
Er lijkt bij elke meting echt NULL binnen te komen continue (volgens de database althans), als je dan vervolgens naar de data zelf kijkt klopt het.

Kan het zijn dat ie twee dataloggers probeert uit te lezen en één v/d twee niet werkt waardoor je NULL waardes krijgt? Ik zou dan meer dan 1 row met null waardes verwachten.

dennissiemensma, 2022-04-24:
Je kunt kijken of je wat wijzer wordt met debug mode. Ik weet niet hoe het via jouw setup moet, maar wellicht is een envvar + herstart voldoende. Je zou dan in de logs ook moeten zien wat de datalogger uitleest en qua backend proces meer informatie onderwater wat die probeert te doen en wat er dan faalt. https://dsmr-reader.readthedocs.io/en/v4/how-to/troubleshooting/enabling-debug-logging.html

randommen96, 2022-04-26:
`2022-04-26 20:19:11,839 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440195) to be marked published by broker
2022-04-26 20:19:11,840 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 184)
2022-04-26 20:19:11,840 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 184)
2022-04-26 20:19:11,841 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440195) to be marked published by broker
2022-04-26 20:19:11,843 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440195) to be marked published by broker
2022-04-26 20:19:11,844 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 184)
2022-04-26 20:19:11,844 DEBUG    broker       run                             106 | MQTT: Deleting published message (#64440195) from queue
2022-04-26 20:19:11,850 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64440194) for dsmr/reading/phase_currently_delivered_l3: 0.015
2022-04-26 20:19:11,851 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m185), 'b'dsmr/reading/phase_currently_delivered_l3'', ... (5 bytes)
2022-04-26 20:19:11,853 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440194) to be marked published by broker
2022-04-26 20:19:11,854 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 185)
2022-04-26 20:19:11,854 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 185)
2022-04-26 20:19:11,855 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440194) to be marked published by broker
2022-04-26 20:19:11,856 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 185)
2022-04-26 20:19:11,858 DEBUG    broker       run                             106 | MQTT: Deleting published message (#64440194) from queue
2022-04-26 20:19:11,864 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64440193) for dsmr/reading/phase_currently_delivered_l2: 0.000
2022-04-26 20:19:11,865 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m186), 'b'dsmr/reading/phase_currently_delivered_l2'', ... (5 bytes)
2022-04-26 20:19:11,866 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440193) to be marked published by broker
2022-04-26 20:19:11,867 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 186)
2022-04-26 20:19:11,868 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 186)
2022-04-26 20:19:11,870 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440193) to be marked published by broker
2022-04-26 20:19:11,871 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 186)
2022-04-26 20:19:11,872 DEBUG    broker       run                             106 | MQTT: Deleting published message (#64440193) from queue
2022-04-26 20:19:11,881 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64440192) for dsmr/reading/phase_currently_delivered_l1: 0.098
2022-04-26 20:19:11,882 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m187), 'b'dsmr/reading/phase_currently_delivered_l1'', ... (5 bytes)
2022-04-26 20:19:11,883 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440192) to be marked published by broker
2022-04-26 20:19:11,884 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 187)
2022-04-26 20:19:11,885 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 187)
2022-04-26 20:19:11,886 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440192) to be marked published by broker
2022-04-26 20:19:11,887 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440192) to be marked published by broker
2022-04-26 20:19:11,888 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 187)
2022-04-26 20:19:11,889 DEBUG    broker       run                             106 | MQTT: Deleting published message (#64440192) from queue
2022-04-26 20:19:11,896 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64440191) for dsmr/reading/electricity_currently_returned: 0.000
2022-04-26 20:19:11,896 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m188), 'b'dsmr/reading/electricity_currently_returned'', ... (5 bytes)
2022-04-26 20:19:11,898 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64440191) to be marked published by broker
[... cut ...]
2022-04-26 20:21:00,629 DEBUG    messages     queue_message                    42 | MQTT: Queued message for dsmr/meter-stats/voltage_swell_count_l3: 1
2022-04-26 20:21:00,640 DEBUG    messages     queue_message                    42 | MQTT: Queued message for dsmr/meter-stats/rejected_telegrams: 35607
2022-04-26 20:21:00,653 DEBUG    mixins       run_loop                         75 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s`

Zojuist even geprobeerd, maar zie niet direct iets heel vreemds gebeuren, jij wel?

dennissiemensma, 2022-04-26:
Bedankt voor je update. Ik zoek twee dingen: telegram en fouten. 

Het telegram staat er in en lijkt verder gewoon goede waardes te hebben, wat NULL values alleen maar vreemder maken.
Voor fouten (als ze er zijn) kun je wellicht wat ruis wegnemen door de logs te catten of tailen met `grep -v 'MQTT:'`, zodat een hoop van de MQTT spam verdwijnt.

In principe komt er elke X seconden (of minuut) een proces langs in die backend die waardes insert in de `dsmr_consumption_electricityconsumption`, ik ben wel benieuwd of daar iets voor langs komt als je een minuutjes zoekt/filtert zonder MQTT.

dennissiemensma, 2022-04-26:
Deze zoek ik: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmrreader/config/defaults.py#L47

In de logs kan die wel iets als `Generate consumption data` heten: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_consumption/migrations/0013_schedule_generate_consumption.py#L10

randommen96, 2022-04-26:
Zoiets?

```@pi01:/home/pi# tail -f /var/log/syslog | grep 'consumption' | grep -v 'MQTT:'
Apr 26 20:55:33 pi01 f25b13ef9683[853]: 2022-04-26 20:55:33,029 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Generate consumption data" to 2022-04-26 18:55:34.008379+00:00 (ETA 0:00:00.979463)
Apr 26 20:57:43 pi01 f25b13ef9683[853]: 2022-04-26 20:57:43,094 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)

@pi01:/home/pi# cat /var/log/syslog | grep 'consumption' | grep '20:57' | grep -v 'MQTT'
Apr 26 09:20:57 pi01 87dbba4bb308[853]: 2022-04-26 07:20:57.654 UTC [8603] ERROR:  null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 09:20:57 pi01 87dbba4bb308[853]: 2022-04-26 07:20:57.654 UTC [8603] STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2022-04-01T19:24:00+00:00'::timestamptz, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"
Apr 26 09:20:57 pi01 5aedeeba11ca[853]: 2022-04-26 09:20:57,660 ERROR    schedule     execute_scheduled_processes      37 | (IntegrityError) dsmr_consumption.services.run errored: null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 20:57:43 pi01 f25b13ef9683[853]: 2022-04-26 20:57:43,094 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)```

randommen96, 2022-04-26:
Dit ziet er wel interessant uit:

```@pi01:/home/pi# cat /var/log/syslog | grep '20:57' | grep -v 'MQTT'
Apr 26 06:20:57 pi01 5aedeeba11ca[853]: 127.0.0.1 - - [26/Apr/2022:06:20:57 +0200] "GET /about HTTP/1.1" 200 16244 "-" "curl/7.80.0" "-"
Apr 26 09:20:57 pi01 87dbba4bb308[853]: 2022-04-26 07:20:57.654 UTC [8603] ERROR:  null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 09:20:57 pi01 87dbba4bb308[853]: 2022-04-26 07:20:57.654 UTC [8603] DETAIL:  Failing row contains (610973, 2022-04-01 19:24:00+00, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
Apr 26 09:20:57 pi01 87dbba4bb308[853]: 2022-04-26 07:20:57.654 UTC [8603] STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2022-04-01T19:24:00+00:00'::timestamptz, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"
Apr 26 09:20:57 pi01 5aedeeba11ca[853]: 2022-04-26 09:20:57,660 ERROR    schedule     execute_scheduled_processes      37 | (IntegrityError) dsmr_consumption.services.run errored: null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 09:20:57 pi01 5aedeeba11ca[853]: DETAIL:  Failing row contains (610973, 2022-04-01 19:24:00+00, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
Apr 26 09:20:57 pi01 5aedeeba11ca[853]:
Apr 26 11:20:57 pi01 5aedeeba11ca[853]: 127.0.0.1 - - [26/Apr/2022:11:20:57 +0200] "GET /about HTTP/1.1" 200 16244 "-" "curl/7.80.0" "-"
Apr 26 15:20:57 pi01 5aedeeba11ca[853]: 127.0.0.1 - - [26/Apr/2022:15:20:57 +0200] "GET /about HTTP/1.1" 200 16245 "-" "curl/7.80.0" "-"
Apr 26 20:20:57 pi01 bfef307c58d8[853]: 2022-04-26 20:20:57,987 DEBUG    schedule     execute_scheduled_processes      29 | SP: 0 backend service(s) ready to run
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,352 DEBUG    services     compact                          76 | Compact: Processed reading: 5636255 @ 2022-04-25 20:57:00+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,386 DEBUG    services     compact                          76 | Compact: Processed reading: 5636256 @ 2022-04-25 20:57:06+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,417 DEBUG    services     compact                          76 | Compact: Processed reading: 5636257 @ 2022-04-25 20:57:12+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,450 DEBUG    services     compact                          76 | Compact: Processed reading: 5636258 @ 2022-04-25 20:57:18+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,482 DEBUG    services     compact                          76 | Compact: Processed reading: 5636259 @ 2022-04-25 20:57:24+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,515 DEBUG    services     compact                          76 | Compact: Processed reading: 5636260 @ 2022-04-25 20:57:30+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,546 DEBUG    services     compact                          76 | Compact: Processed reading: 5636261 @ 2022-04-25 20:57:36+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,583 DEBUG    services     compact                          76 | Compact: Processed reading: 5636262 @ 2022-04-25 20:57:42+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,636 DEBUG    services     compact                          76 | Compact: Processed reading: 5636263 @ 2022-04-25 20:57:48+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,668 DEBUG    services     compact                          76 | Compact: Processed reading: 5636264 @ 2022-04-25 20:57:54+02:00
Apr 26 20:43:07 pi01 bfef307c58d8[853]: 2022-04-26 20:43:07,417 DEBUG    services     compact                          76 | Compact: Processed reading: 5642483 @ 2022-04-26 07:20:57+02:00
Apr 26 20:54:36 pi01 f25b13ef9683[853]: 2022-04-26 20:54:36,655 DEBUG    services     compact                          76 | Compact: Processed reading: 5647274 @ 2022-04-26 15:20:57+02:00
Apr 26 20:57:00 pi01 f25b13ef9683[853]: 2022-04-26 20:57:00,123 DEBUG    mixins       run_loop                         75 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
Apr 26 20:57:00 pi01 f25b13ef9683[853]: 2022/04/26 20:57:00 [error] 417#417: *88 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
Apr 26 20:57:00 pi01 f25b13ef9683[853]: 127.0.0.1 - - [26/Apr/2022:20:57:00 +0200] "GET /about HTTP/1.1" 504 160 "-" "curl/7.80.0" "-"
[... cut ...] s': 0}
Apr 26 20:57:05 pi01 f25b13ef9683[853]: 2022-04-26 20:57:05,495 DEBUG    dsmr_datalogger_api_client read_telegram                    54 | [2022-04-26 20:57:05.494961] Read 1 Byte(s)
Apr 26 20:57:05 pi01 f25b13ef9683[853]: 2022-04-26 20:57:05,830 DEBUG    dsmr_datalogger_api_client read_telegram                    54 | [2022-04-26 20:57:05.830659] Read 913 Byte(s)
Apr 26 20:57:05 pi01 f25b13ef9683[853]: 2022-04-26 20:57:05,849 DEBUG    datalogger   telegram_to_reading              75 | ```

dennissiemensma, 2022-04-26:
Hij **lijkt** overigens verder wel door te lopen als ik dit zo zie:
```
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,352 DEBUG    services     compact                          76 | Compact: Processed reading: 5636255 @ 2022-04-25 20:57:00+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,386 DEBUG    services     compact                          76 | Compact: Processed reading: 5636256 @ 2022-04-25 20:57:06+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,417 DEBUG    services     compact                          76 | Compact: Processed reading: 5636257 @ 2022-04-25 20:57:12+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,450 DEBUG    services     compact                          76 | Compact: Processed reading: 5636258 @ 2022-04-25 20:57:18+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,482 DEBUG    services     compact                          76 | Compact: Processed reading: 5636259 @ 2022-04-25 20:57:24+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,515 DEBUG    services     compact                          76 | Compact: Processed reading: 5636260 @ 2022-04-25 20:57:30+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,546 DEBUG    services     compact                          76 | Compact: Processed reading: 5636261 @ 2022-04-25 20:57:36+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,583 DEBUG    services     compact                          76 | Compact: Processed reading: 5636262 @ 2022-04-25 20:57:42+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,636 DEBUG    services     compact                          76 | Compact: Processed reading: 5636263 @ 2022-04-25 20:57:48+02:00
Apr 26 20:29:48 pi01 bfef307c58d8[853]: 2022-04-26 20:29:48,668 DEBUG    services     compact                          76 | Compact: Processed reading: 5636264 @ 2022-04-25 20:57:54+02:00
```

Of herstart/bump je hem telkens?

dennissiemensma, 2022-04-26:
Oh wacht, nvm, de telegrammen zijn van een dag eerder

randommen96, 2022-04-26:
Klopt inderdaad, had momenteel NULL values toegestaan in de database zodat het even kon doorlopen en de queue weer verdween

Hieronder zie je een insert statement waar het fout gaat (even zonder code blocks, anders verdwijnt om de een of andere reden de opmaak en is het vrij onleesbaar), kun je hier wat mee?
Apr 26 20:28:22 pi01 87dbba4bb308[853]: 2022-04-26 18:28:22.314 UTC [21439] ERROR:  null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 20:28:22 pi01 87dbba4bb308[853]: 2022-04-26 18:28:22.314 UTC [21439] DETAIL:  Failing row contains (612261, 2022-04-01 19:24:00+00, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
Apr 26 20:28:22 pi01 87dbba4bb308[853]: 2022-04-26 18:28:22.314 UTC [21439] STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2022-04-01T19:24:00+00:00'::timestamptz, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,321 ERROR    schedule     execute_scheduled_processes      37 | (IntegrityError) dsmr_consumption.services.run errored: null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 26 20:28:22 pi01 bfef307c58d8[853]: DETAIL:  Failing row contains (612261, 2022-04-01 19:24:00+00, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
Apr 26 20:28:22 pi01 bfef307c58d8[853]:
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,329 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Generate consumption data" to 2022-04-26 18:28:52.322323+00:00 (ETA 0:00:29.993313)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,345 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,352 DEBUG    broker       run                              76 | MQTT: Processing 33 message(s)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,353 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64443220) for dsmr/meter-stats/rejected_telegrams: 35607
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,354 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m2960), 'b'dsmr/meter-stats/rejected_telegrams'', ... (5 bytes)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,356 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64443220) to be marked published by broker
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,357 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 2960)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,358 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 2960)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,360 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64443220) to be marked published by broker
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,361 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 2960)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,364 DEBUG    broker       run                             106 | MQTT: Deleting published message (#64443220) from queue
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,373 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#64443219) for dsmr/meter-stats/voltage_swell_count_l3: 1
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,374 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m2961), 'b'dsmr/meter-stats/voltage_swell_count_l3'', ... (1 bytes)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,377 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64443219) to be marked published by broker
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,378 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 2961)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,379 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 2961)
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,381 DEBUG    broker       run                              98 | MQTT: Waiting for message (#64443219) to be marked published by broker
Apr 26 20:28:22 pi01 bfef307c58d8[853]: 2022-04-26 20:28:22,383 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 2961)

dennissiemensma, 2022-04-27:
Het mechanisme is namelijk vrij simpel. De records in de tabel met metingen worden gesplitst naar aparte tabellen voor electriciteit en gas: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_consumption/services.py#L62-L71

- Als je de metingen niet groepeert dan is het bijna copy-paste van de metingentabel: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_consumption/services.py#L91
- Als je metingen per minuut groepeert, pakt die gewoon de hoogste waarde van die minuut: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_consumption/services.py#L129

Heb je trouwens DSMR-reader en de database wel in dezelfde tijdzone staan? Het staat mij niet bekend als oorzaak voor jouw issue, maar daar kunnen altijd wel vage dingen door naar voren komen. Er zijn ook wel eens issues geweest met bepaalde packages.

dennissiemensma, 2022-04-27:
Sterker nog! Nu ik dat tijdzone-issue weer terug heb gezocht, zie ik wel degelijk dezelfde foutmelding!

# #1282
```
dsmrdb_1  | 2021-02-10 22:29:00.344 CET [38] ERROR:  null value in column "delivered_1" violates not-null constraint
dsmrdb_1  | 2021-02-10 22:29:00.344 CET [38] DETAIL:  Failing row contains (70779, 2021-02-09 04:00:00+01, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
dsmrdb_1  | 2021-02-10 22:29:00.344 CET [38] STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2021-02-09T03:00:00+00:00'::timestamptz, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"
```

Wil je eens kijken of het mogelijk daar mee te maken kan hebben? Het is echter geen recent probleem, dus ik weet niet zeker of het dezelfde oorzaak is.

randommen96, 2022-04-27:
Tijdzone gecheckt, in de database en dsmr container via "date" en die staat goed, ook de postgres.conf nog gecheckt, die stond op UTC, aangepast naar Europe/Amsterdam, zijn er nog meer plekken die ik na moet lopen? Kan me zo even niets bedenken.

Wat me wel direct opvalt, als ik naar de live telegram readings ga onder -> http://url:7777/admin/dsmr_datalogger/meterstatistics/ dan zie je in de "MeterStatistics @ 2022-04-27 10:48:26+00:00" een tijd van twee uur terug, maar vervolgens heeft de timestamp wel weer de jusite tijd "Timestamp:
April 27, 2022, 12:48 p.m.
Timestamp indicating when the reading was taken" Hoort dat zo?

De select query die je gaf kwam niet terug met resultaten, wat me daarentegen wel opvalt, na even wat gekeken en gerommeld te hebben, is dat hij lijkt te hangen op een INSERT van 1 april (dit is tegelijkertijd ook wat je voor de INSERT ziet gebeuren zoals je vroeg):
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,070 DEBUG    mixins       run_loop                         75 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,399 DEBUG    schedule     execute_scheduled_processes      29 | SP: 1 backend service(s) ready to run
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,399 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,476 ERROR    schedule     execute_scheduled_processes      37 | (IntegrityError) dsmr_consumption.services.run errored: null value in column "delivered_1" of relation "dsmr_consumption_electricityconsumption" violates not-null constraint
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: DETAIL:  Failing row contains (614728, 2022-04-01 19:24:00+00, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]:
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,487 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Generate consumption data" to 2022-04-27 10:44:30.477472+00:00 (ETA 0:00:29.990580)
Apr 27 12:44:00 pi01 e1cf8bcecd90[853]: 2022-04-27 12:44:00,507 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)

Als ik het trucje uithaal, maakt niet uit hoe vaak, door de NOT NULL te droppen en daarna de DELETE en dan weer NULL te weigeren, blijft deze INSERT statement terugkomen en ook telkens een NULL rij, heb jij een idee hoe ik die INSERT er "tussenuit" krijg? Ik denk dat we er dan misschien wel zijn? Ik snap alleen niet hoe die dan ontstaan is, behalve dan dat ik een restore heb gedaan een week of twee geleden na de problemen met TimescaleDB.

dennissiemensma, 2022-04-28:
> dan zie je in de "MeterStatistics @ 2022-04-27 10:48:26+00:00" een tijd van twee uur terug, maar vervolgens heeft de timestamp wel weer de jusite tijd "Timestamp: April 27, 2022, 12:48 p.m. Timestamp indicating when the reading was taken" 
> Hoort dat zo?

Jazeker, want het hangt af van de tijdzone offset die erbij zit: `10:48:26+00:00`, waarbij de `+00:00` aangeeft dat het UTC is. UTC is in onze wintertijd 1 uur eerder en in de zomertijd (dus nu) 2 uur eerder. 
Als je van `10:48:26+00:00` UTC onze tijd maakt, is het ``12:48:26+02:00`` Nederlandse zomertijd.

----

Ik zie trouwens dat mn [eerdere query](https://github.com/dsmrreader/dsmr-reader/issues/1619#issuecomment-1110724499) niet klopte. Wil je deze eens proberen?

```
SELECT timestamp, electricity_delivered_1, electricity_delivered_2 FROM dsmr_datalogger_dsmrreading WHERE processed = false ORDER BY timestamp ASC;
```

Daarmee zie je alle onverwerkte metingen. Zie je daar wellicht NULL waardes?

randommen96, 2022-04-28:
Niet dat mij opvalt, jou wel?
[image]
Na het verwerken van alle metingen, blijven die van 1 april staan, verder verwerkt hij netjes alle nieuwe metingen, totdat ik de NULL constraint weer aanzet, had hem weer even uit omdat er 15k in de backlog stonden.

Wellicht die van 1 april eruit mieteren?

Mooi dat de tijd klopt en erg helder uitgelegd, dank daaroor :-)

dennissiemensma, 2022-04-28:
Je kunt dan deze query doen:

```
SELECT id, timestamp, electricity_delivered_1, electricity_delivered_2 FROM dsmr_datalogger_dsmrreading WHERE processed = false ORDER BY timestamp ASC;
```

En dan de 3 ID's die bij die records horen verwijderen, bijvoorbeeld ID's 1111, 2222 en 3333:
```
DELETE FROM dsmr_datalogger_dsmrreading WHERE id = 1111;
DELETE FROM dsmr_datalogger_dsmrreading WHERE id = 2222;
DELETE FROM dsmr_datalogger_dsmrreading WHERE id = 3333;
```

dennissiemensma, 2022-04-29:
Ik weet niet hoe pgadmin meerdere queries afhandelt, maar als die je 3 deletes los van elkaar doet, zou het sowieso moeten werken.

Verder krijg je op je SELECT-query geen resultaten omdat je geen velden opgeeft. Ik wist niet dat dat uberhaupt werkte.

Probeer maar iets als:
```
SELECT id, timestamp FROM dsmr_datalogger_dsmrreading WHERE id = 5686383;
```

randommen96, 2022-04-30:
Ben even aan het spelen geweest, er lijkt iets vreemds te gebeuren in postgres zelf en ik ben er nog niet achter waardoor...
Doe ik een enkele SELECT of DELETE, krijg ik 0 rows terug, echter gebeurt dat alleen met de 3 id's van 1 april.

Doe ik een SELECT op een willekeurige daarna, dan krijg ik netjes de row terug zoals verwacht:
[image]

Doe je het op eentje van 1 april:
[image]

Er moet dus haast wel iets mis zijn met die 3 rows, ik krijg ze alleen écht niet weg:
[image]

Ook als je de pgadmin delete functie gebruikt, verdwijnen ze wel op het oog, doe je daarna weer alle unprocessed ophalen dan staan ze er nog (ook met DSMR reader uit, dus het zit echt in postgres), heel vreemd.

dennissiemensma, 2022-04-30:
Je kunt het nog buiten pgadmin proberen, direct op de command line bij de database via `psql`?  Qua syntax zijn de queries gelijk, gebruik is zoiets: https://github.com/dsmrreader/dsmr-reader/issues/1551#issuecomment-1007938922

En ik neem aan dat je de DB ook al een keer herstart hebt?

randommen96, 2022-04-30:
psql had ik inderdaad ook geprobeerd en heb meerdere keren herstart ja.

Heb het eindelijk opgelost, besloten om de tabel te TRUNCATEN en weer opnieuw in te laden met:
BEGIN;
COPY public.dsmr_datalogger_dsmrreading TO '/tmp/dsmr_datalogger_dsmrreading.data';
TRUNCATE TABLE public.dsmr_datalogger_dsmrreading;
COPY public.dsmr_datalogger_dsmrreading FROM '/tmp/dsmr_datalogger_dsmrreading.data';
END;

Toen kwam naar voren, bij het importeren, dat er sprake was van duplicate primary keys (id's),
na een aantal lines handmatig verwijderd te hebben bleven ze maar komen, besloten om het op te lossen door een klein python scriptje te maken:
```
import hashlib

output_file_path = "/mnt/c/Users/David/Downloads/dsmrreader/dsmr_out.txt"
input_file_path = "/mnt/c/Users/David/Downloads/dsmrreader/dsmr_in.txt"

completed_ids_hash = set()

output_file = open(output_file_path, "w")

for line in open(input_file_path, "r"):
  rowid = line.split()[0]
 
  hashValue = hashlib.md5(rowid.rstrip().encode('utf-8')).hexdigest()
 
  if hashValue not in completed_ids_hash:
    output_file.write(line)
    completed_ids_hash.add(hashValue)
  else:
    print(rowid)

output_file.close()
```

Vervolgens de tabel succesvol teruggezet weer met:
COPY public.dsmr_datalogger_dsmrreading FROM '/tmp/dsmr_datalogger_dsmrreading.data';

Nog eenmaal gekeken met pgadmin of de onverwerkte 1 april telegrams er nog stonden...
Deze stonden er nog, nu succesvol kunnen verwijderen en ook écht weg.

Uiteindelijk dsmr reader weer gestart en de onverwerkte telegrams weg laten werken, de tabel loopt nu netjes door zoals het zou moeten.

Wat alleen wel opvalt in de DEBUG log nu is het volgende:
```
Apr 30 13:18:55 pi01 d37d65340783[853]: 2022-04-30 13:18:55,290 DEBUG    schedule     execute_scheduled_processes      29 | SP: 1 backend service(s) ready to run
Apr 30 13:18:55 pi01 d37d65340783[853]: 2022-04-30 13:18:55,290 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
Apr 30 13:18:55 pi01 d37d65340783[853]: 2022-04-30 13:18:55,660 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 13:18:55 pi01 d37d65340783[853]: 2022-04-30 13:18:55,798 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
Apr 30 13:18:56 pi01 systemd[5052]: run-docker-runtime\x2drunc-moby-d37d653407830fc790f5e5ed53841d29120114bfe9f7367de3edc8db431ccb93-runc.9GaRoG.mount: Succeeded.
Apr 30 13:18:56 pi01 systemd[1]: run-docker-runtime\x2drunc-moby-d37d653407830fc790f5e5ed53841d29120114bfe9f7367de3edc8db431ccb93-runc.9GaRoG.mount: Succeeded.
Apr 30 13:18:56 pi01 d37d65340783[853]: 2022-04-30 13:18:56,515 DEBUG    mixins       run_loop                         75 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
Apr 30 13:18:56 pi01 d37d65340783[853]: 127.0.0.1 - - [30/Apr/2022:13:18:56 +0200] "GET /about HTTP/1.1" 200 16247 "-" "curl/7.80.0" "-"
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,526 DEBUG    schedule     execute_scheduled_processes      29 | SP: 3 backend service(s) ready to run
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,527 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Generate day and hour statistics" (dsmr_stats.services.run)
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,632 DEBUG    services     create_daily_statistics         133 | Stats: Creating day statistics for: 2022-04-01
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,667 ERROR    schedule     execute_scheduled_processes      37 | (TypeError) dsmr_stats.services.run errored: unsupported operand type(s) for -: 'NoneType' and 'decimal.Decimal'
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,675 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Generate day and hour statistics" to 2022-04-30 11:19:27.668393+00:00 (ETA 0:00:29.992892)
Apr 30 13:18:57 pi01 d37d65340783[853]: 2022-04-30 13:18:57,676 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,481 DEBUG    services     compact                          59 | Compact: Waiting for newer readings before grouping data...
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,489 DEBUG    schedule     reschedule                       75 | SP: Rescheduled "Generate consumption data" to 2022-04-30 11:19:13.481780+00:00 (ETA 0:00:14.992213)
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,490 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,843 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,981 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
Apr 30 13:18:58 pi01 d37d65340783[853]: 2022-04-30 13:18:58,993 DEBUG    mixins       run_loop                         75 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[... cut ...]
Apr 30 13:19:00 pi01 d37d65340783[853]: 2022-04-30 13:19:00,500 DEBUG    dsmr_datalogger_api_client read_telegram                    54 | [2022-04-30 13:19:00.499921] Read 914 Byte(s)
```

En dan specifiek deze regel:
execute_scheduled_processes      37 | (TypeError) dsmr_stats.services.run errored: unsupported operand type(s) for -: 'NoneType' and 'decimal.Decimal'

Ik vraag me af waar die nu vandaan komt, hij loopt nu dus stuk op 1 april als die de dag statistieken wilt opmaken lijkt het, eens kijken in de tabellen...

randommen96, 2022-04-30:
Kan het komen doordat ie een resultaat zoekt van elk uur maar deze niet kan vinden?
[image]

dennissiemensma, 2022-04-30:
Bedankt voor je aanvulling. Het is helaas wat lastig te duiden, gezien de foutmelding niet de locatie aangeeft. Voor de volgende release zal ik de stacktrace toevoegen, zodat duidelijk is waar die vandaan komt. 

Het stukje code dat de statistieken genereert heeft namelijk diverse plekken waar dit fout kan gaan. Ik denk echter wel dat het inderdaad veroorzaakt kan worden door het missen van een paar uur aan data.

Is de achterstand van je onverwerkte metingen nu wel opgelost? 

Want dan zou voor nu een oplossing zijn om handmatig nog 2 (of 4 als je ook gas hebt) records toe te voegen in de brontabel voor dagstatistieken (voor elk missend uur). Dat is een andere tabel, namelijk `dsmr_consumption_electricityconsumption` en `dsmr_consumption_gasconsumption`. 
Je kunt de waardes eventueel baseren op de eerste waardes van 2 april (een paar uur later), zodat het dagtotaal gelijk is aan de start van 2 april. 
Wellicht kun je via PG-admin zelfs makkelijk die rijen klonen. Voor beide tabellen is er dus per uur 1 record nodig voor 22:00-23:00 en 23:00 - 0:00.

randommen96, 2022-04-30:
Yes metingen lopen nu door dus dat is opgelost :-D

De statistieken worden nu ook weer gegenereerd, er stond een row met null waardes op 1 april, deze eruit gehaald + voor 22:30 en 23:30 een waarde toegevoegd met de queries hieronder, was net iets makkelijker dan via PG-Admin, en dsmr reader gestart en hij is netjes doorgestroomd.

Die twee waardes toevoegen was waarschijnlijk niet eens nodig, ben dit vergeten bij de gas tabel, en die staat ook netjes in de dagstatistieken + ik mis de data tussen 1 en 7 april, omdat ik op 7 april de backup heb teruggezet. Dat er een null row in die tabel stond kan ook prima kloppen, er zit namelijk geen constraint op die tabel zie ik, verklaard ook de Nonetype error.

```
SELECT * FROM dsmr_consumption_electricityconsumption WHERE read_at::date = date '2022-04-01' ORDER BY read_at ASC;

INSERT INTO dsmr_consumption_electricityconsumption ( read_at , delivered_1, returned_1, delivered_2, returned_2, currently_delivered, currently_returned, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3, phase_power_current_l1, phase_power_current_l2, phase_power_current_l3)
SELECT '2022-04-01 23:30:00+02', delivered_1, returned_1, delivered_2, returned_2, currently_delivered, currently_returned, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3, phase_power_current_l1, phase_power_current_l2, phase_power_current_l3
FROM dsmr_consumption_electricityconsumption
WHERE id = 546406;
```
(id aanpassen aan een rij die je wilt kopieren, postgres creeert zelf een unieke ID, maak ook de timestamp uniek. als je dat vergeet krijg je een error en draait de query niet)

Er gaat nu nog één ding fout... dat is de data rotation, alleen die Stacktrace is nog kleiner dus de debug data zegt mij niet zoveel, wat opvalt is dat het volgende continue voorkomt:
```
Apr 30 21:54:52 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:54:52,801 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 21:54:53 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:54:53,962 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
Apr 30 21:54:54 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:54:54,338 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 21:54:55 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:54:55,492 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
```

en met 10 regels ervoor en erna ziet dat er zo uit:
```
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 1-0:41.7.0(00.033*kW)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 1-0:61.7.0(00.142*kW)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 1-0:22.7.0(00.000*kW)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 1-0:42.7.0(00.000*kW)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 1-0:62.7.0(00.000*kW)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 0-1:24.1.0(003)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 0-1:96.1.0(4730303738353635353837373735333230)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: 0-1:24.2.1(220430215500S)(03299.632*m3)#015
Apr 30 21:55:49 pi01 c78e8b9e8a3f[853]: !BFB1
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:50,101 DEBUG    schedule     execute_scheduled_processes      29 | SP: 1 backend service(s) ready to run
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:50,101 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:50,239 DEBUG    mixins       run_loop                         75 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
Apr 30 21:55:50 pi01 systemd[22307]: run-docker-runtime\x2drunc-moby-c78e8b9e8a3f4751ae2783b429e006f70848c778eaacd1a7ce922c79203ba40c-runc.gdscBu.mount: Succeeded.
Apr 30 21:55:50 pi01 systemd[1]: run-docker-runtime\x2drunc-moby-c78e8b9e8a3f4751ae2783b429e006f70848c778eaacd1a7ce922c79203ba40c-runc.gdscBu.mount: Succeeded.
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 127.0.0.1 - - [30/Apr/2022:21:55:50 +0200] "GET /about HTTP/1.1" 200 15655 "-" "curl/7.80.0" "-"
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:50,541 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 21:55:50 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:50,695 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
Apr 30 21:55:51 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:51,257 DEBUG    mixins       run_loop                         75 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
Apr 30 21:55:52 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:52,264 DEBUG    schedule     execute_scheduled_processes      29 | SP: 2 backend service(s) ready to run
[... cut ...]
Apr 30 21:55:53 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:53,084 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
Apr 30 21:55:53 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:53,430 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
Apr 30 21:55:53 pi01 c78e8b9e8a3f[853]: 2022-04-30 21:55:53,568 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
```

Zegt jou dit nog iets waar ik kan zoeken?

dennissiemensma, 2022-04-30:
Het zit in het mechanisme van opschoning. Wellicht komt die ook niet verder als er data mist, maar dat kan ik niet zo 1-2-3 zien of zeggen.

De logica is dit: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_datalogger/services/retention.py#L39
Het komt er op neer dat die zoekt naar uren met meer dan 2 records (3 tabellen worden opgeschoond) en dan verwijdert die per uur alle records, behalve de eerste en laatste van een uur. 

Effectief krijg je dan wat je op een van je eerdere screenshots ook al zag: https://github.com/dsmrreader/dsmr-reader/issues/1619#issuecomment-1113971550

randommen96, 2022-05-01:
Ff aantal dingen geprobeerd, eerst data toegevoegd zoals je zei, dus op minuut 00 en 59 van elk uur, dat hielp al iets:
```
@pi01:/home/pi# tail -f /var/log/syslog | grep -v 'MQTT' | grep 496e9a4b0d95 | grep retent
May  1 20:29:04 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:04,502 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
May  1 20:29:04 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:04,988 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
May  1 20:29:05 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:05,033 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 20:00:00+00:00 (ElectricityConsumption)
May  1 20:29:05 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:05,077 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 21:00:00+00:00 (ElectricityConsumption)
May  1 20:29:07 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:07,091 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
May  1 20:29:07 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:07,451 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
May  1 20:29:08 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:08,607 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
May  1 20:29:09 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:09,024 DEBUG    retention    run                              76 | Retention: Cleaning up: 2022-04-01 19:00:00+00:00 (ElectricityConsumption)
May  1 20:29:10 pi01 496e9a4b0d95[853]: 2022-05-01 20:29:10,208 DEBUG    schedule     execute_scheduled_processes      32 | SP: Running "Retention data rotation" (dsmr_datalogger.services.retention.run)
```
Zoals je ziet verwerkt hij netjes de toegevoegde uren, en valt dan vervolgens weer in de herhaling,
toen zag ik dat er na het cleanen om 19:00 UTC, dus 21:00 onze tijd, nog data in stond buiten de 00 en 59 minuut, alles ertussen weggegooid toen DSMR reader uitstond, weer gestart, en nu gaat ie door.

Eind goed al goed, zo zien we het graag:
[image]

:-D

Thanks voor alle hulp, veel geleerd en ik hoop dat het ook nog ergens meehelpt aan de ontwikkeling ooit, hoewel het hier uiteindelijk om een stukje data corruptie leek te gaan (dubbele primary keys e.d.), de vraag is natuurlijk hoe dicht je het wilt timmeren qua failsafe.

Sluit jij de issue als je tevreden bent of eventueel nog iets van debug informatie wilt hebben van mij?

dennissiemensma, 2022-05-01:
Bedankt voor je update en fijn dat het uiteindelijk gelukt is!

Het helpt zeker mee, want voor de volgende release zal ik het debuggen wat makkelijker maken met betere stacktraces en optionele query logging.
De dataverwerking zelf kan uiteraard ook nog wel wat beter, alleen is het dusdanig veel werk om alle uitzonderingen af te vangen, dat ik dat meestal pas doe als een specifiek probleem vaak gemeld wordt. Sowieso zijn uitgebreide issues als deze altijd handig als naslagwerk voor mij of iemand anders!

closed 2022-05-01 by dennissiemensma

poudenes, 2022-06-14:
Hoi Allen,

Ik zit eigenlijk met hetzelfde. Alleen begrijp ik niet wat hier allemaal gezegd is en is uitgevoerd.
Ik heb mijn backup van DSMR (gemaakt in postgres/timescale) in een postgres dB terug gezet. 
Alles ging goed. DSMR gestart en alle oude waardes en archief waren terug.

Echter zie ik nu een heleboel telegram regels die niet verwerkt zijn:
```
Processed: False
Geeft aan of deze meting al verwerkt is in het samenvoegen van statistieken
```
Timegap van de backup en start nieuwe dB
```
14 juni 2022 11:31 |   | 7281.856 | 7717.392 | - | - | 2453.176
-- | -- | -- | -- | -- | -- | --
  |   | 7279.459 | 7713.720 | - | -

[14 juni 2022 11:31](http://192.168.100.24:8123/api/hassio_ingress/W7EgxaPxvaMf_GoMbJcwoLE7w2pKvz7Oahk_dORLOGA/admin/dsmr_datalogger/dsmrreading/22022661/change/)	False	7281.856	7717.392	-	-	2453.176
	[13 juni 2022 15:12](http://192.168.100.24:8123/api/hassio_ingress/W7EgxaPxvaMf_GoMbJcwoLE7w2pKvz7Oahk_dORLOGA/admin/dsmr_datalogger/dsmrreading/22022557/change/)	True	7279.459	7713.720	-	-
```

dennissiemensma, 2022-06-14:
@poudenes je kunt het beste een nieuw issue openen en evt verwijzen naar deze, om te voorkomen dat anderen hier ook allemaal notificaties van krijgen. 
Daarnaast komt het regelmatig voor dat het per gebruiker een andere oorzaak heeft, wat het alleen maar verwarrender maakt voor anderen die dit ooit teruglezen.

Voeg, als het kan, in je nieuwe issue ook de informatie toe die gevraagd wordt zoals de debug info dump.