dsmrreader/dsmr-reader #1620: 💡 Request - Publish netto power flow

stefaanbolle, 2022-04-26:
### Feature

Publish netto power flow to MQTT
netto = dsmr/reading/electricity_currently_delivered - dsmr/reading/electricity_currently_returned 
publish netto to i.e. dsmr/reading/electricity_netto
This would indicate whether power is export (-) or import (+) mode and makes it available as a single number to other automation systems.

dennissiemensma, 2022-04-26:
Bedankt voor je verzoek! (ik ga er vanuit dat je Nederlands bent)

Hoewel ik de aard van je verzoek goed begrijp, ben ik minder een voorstander van het aliassen van triviale berekeningen. Zowel de API als MQTT is in feite een data-export zoals de metergegevens (versimpeld) zijn en het is dan aan externe systemen om daar context aan te geven. Als het nou een hele complexe berekening was, dan zou ik het nog overwegen, alleen is dit een relatief makkelijke switch of conditie om op te nemen in een mapper, of waar je het ook uitleest. 

Daarnaast is het goed om te weten dat ik eerder de fout ben ingegaan door [aannames te doen](https://github.com/dsmrreader/dsmr-reader/issues/1324#issuecomment-802318279) op hoe de situaties bij gebruikers thuis zijn. Specifiek voor jouw verzoek geldt wellicht #1324.
DSMR-reader toonde eerst ofwel export ofwel import, omdat ik dacht dat het of het ene of het andere was, maar dat bleek totaal niet zo te zijn in de praktijk. Het is blijkbaar mogelijk dat een slimme meter [tegelijkertijd teruglevert als stroom uit het net haalt](https://github.com/dsmrreader/dsmr-reader/issues/1324#issuecomment-802735955). Dus ook daarom probeer ik dit soort aannames wat weg te houden. 

Ook al kan het in jouw eigen situatie wel weer of-of zijn. Ik heb zelf nog een oude DSMR-4 meter zonder zonnepanelen, dus ik ben behoorlijk afhankelijk van externe input.

To-Tall, 2022-05-02:
Ik zou hem graag weer willen heropenen :) 

Als ik naar mijn eigen situatie kijk, Dan heb ik Home Assistant en Grafana om netto verbruik dus. Geleverd - terug geleverd == netto verbruik te berekeken.  

Nu hebben mijn ouders, schoonouders ook zonnepanelen. Die gebruiken nu DSMR reader om inzicht te krijgen in het verbruik vs teruglevering. En als extra controlle op de app die ze op de telefoon hebben.  

als niet dever kan ik makkelijk praten :+ maar de berekening is niet heel ingewikkeld om te maken.
In het dasboard zou dan een nieu tabel komen. Geleverd, terug geleverd, netto verbruik. meer geleverd dan verbruikt kleurtje groen, minder terug geleverd dan gebruik == blauw.  bijvoorbeeld :)

dennissiemensma, 2022-05-02:
@To-Tall bedankt voor je verzoek. Zie onder andere https://github.com/dsmrreader/dsmr-reader/issues/1620#issuecomment-1110013416.

DSMR-reader probeert zo dicht mogelijk bij de slimme meter data te blijven, omdat die amper verandert. De intepretatie van die data is echter wel aan verandering onderhevig, zoals per 1 juli in Belgie en mogelijk vanaf volgend jaar voor salderen in Nederland ook (afbouwend salderen).