dsmrreader/dsmr-reader #1621: 💡Toevoegen enkele algemene mqtt topics

Giel538, 2022-05-01:
### Feature

Ik gebruik je DSMR reader nog steeds met veel plezier. Vooral ook voor de goede doorvoor via MQTT. 
Ik kijk eigenlijk niet zo heel vaak op de web interface zelf om te kijken of er een update is en omdat ik toch up to date wil blijven was ik benieuwd of je 2 mqtt topics wil toevoegen. Deze kan ik dan zelf in mijn eigen systeem toevoegen (home assistant).

update available (true or false)
current version (bijv. 5.1.2)

Mijn 2e vraag is of het mogelijk is een LWT message toe te voegen. Dat is een  "status" topic met de tekst online of offline.  Volgens mij schrijf je bij het verbinden met on_connect ofzo er de tekst "online" heen. Als voor wat voor reden dan ook MQTT niet meer kan communiceren met DSMR reader veranderd de status automatisch naar offline en zo kan je ook via mqtt getriggerd worden dat er iets mis is. O.a. hier staat wat meer info erover: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

dennissiemensma, 2022-05-01:
Bedankt voor je verzoek! Ik zet het op de backlog om er later nog wat dieper naar te kijken. 

Ik wil de MQTT-integratie/export namelijk ook niet teveel DSMR-reader specifiek maken. De gegevens getoond in DSMR-reader zijn "een" visualisatie van de metergegevens en via MQTT kan iemand (of een ander systeem) daar simpelweg een andere intepretatie van maken. 
Het blijft uiteraard een grijs gebied, want ik begrijp alle use-cases die ik de afgelopen jaren heb gezien.

dennissiemensma, 2022-06-17:
Ik heb hier nog wat dieper naar gekeken. Hoewel ik de use-case begrijp, denk ik dat het iets teveel en te specifiek is. 

MQTT is slechts een relatief klein onderdeel van DSMR-reader. Als het nou de primaire (of enige) taak was van het project (a la dsmr-to-mqtt) dan zou ik het een heel ander verhaal vinden.
_Opzich is laatsgenoemde ook wel een leuk, praktisch en potentieel project. Ware het niet dat ik daar op dit moment geen tijd voor heb, maar wie weet als ik ooit wat meer ruimte heb. Het zou een hoop wensen en overhead voor DSMR-reader schelen (en andersom), omdat zoiets dan puur geent is op het uitlezen van de meter en doorsturen naar MQTT._

closed 2022-06-17 by dennissiemensma

Giel538, 2022-06-17:
Hoi Dennis.

Is prima! Fijn dat je er naar gekeken hebt. 

Een klein stukje code zonder GUI die P1 direct omzet naar MQTT zou best vraag naar zijn. Dit in een docker container en je hoeft er nooit meer naar om te kijken. Als ik zelf tijd heb zou ik met een beetje pik en plak werk het ook nog voor elkaar moeten krijgen maarja de tijd he.... :)  En daarnaast werkt jou dsmr reader prima dus ik zie er ook niet echt reden toe.