dsmrreader/dsmr-reader #1623: Dropbox credentials verwijdert nadat connectie faalt🚨

KobeBergmans, 2022-05-05:
### Description

Ik heb mijn DSMR-reader geconfigureerd om backups uit te voeren naar Dropbox. Het probleem is nu echter dat mijn internetverbinding elke dag een kwartiertje uitvalt (door wat problemen met de router), en wanneer dit gebeurt krijg ik soms deze error:
[image]
Ik denk dus dat dit ligt aan het feit dat er geen internetverbinding is op het moment dat er connectie met Dropbox wordt gezocht.

Het vervelende, wat voor mij een bug lijkt, is dat als gevolg van deze error mijn credentials worden verwijderd en dat ik ze dus telkens opnieuw moet instellen. 

Zou er dan een oplossing zijn om ervoor te zorgen dat deze error er niet voor zorgt dat de credentials verwijderd worden?

### DSMR-reader version

5.1

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.1 / v3.9.7 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 2.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                             828621
    Consumption records electricity / gas (est.)                                                75809 / 42092
```

dennissiemensma, 2022-05-08:
Bedankt voor je melding. Het hangt er vanaf of de SDK van Dropbox dit ondersteunt.

Ik zie in je screenshot een specifieke foutmelding `NewConnectionError`, dus wellicht is daar wat mee te doen.

dennissiemensma, 2022-06-08:
Ik kan het lokaal testen met de envvar:
```
DROPBOX_API_HOST=non-existing-api.localhost
```

dennissiemensma, 2022-06-08:
Ik heb wat rework gedaan voor de volgende release. Mogelijk lost dat het op, we zullen zien.

closed 2022-06-08 by dennissiemensma

KobeBergmans, 2022-06-09:
Dat ziet er goed uit, hartelijk bedankt!

dennissiemensma, 2022-06-09:
Uitgebracht in [v5.3](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.3.0)

KobeBergmans, 2022-06-22:
@dennissiemensma Ik heb het nu ongeveer een week lopen en tot nu toe geen problemen meer dus dat ziet er goed uit. Dank u!