dsmrreader/dsmr-reader #1624: 🚨 psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above

D1StrX, 2022-05-05:
### Description

Nieuwe installatie DSMR met bestaande database gedaan.
Ubuntu 20.04LTS -> Ubuntu 20.22LTS
Python 3.9.9 -> 3.10.4
(interpreteer bovenstaande zoals de distro geïnstalleerd is)

### Issue
Met onderstaande commando erachter gekomen wat het issue is:
 
```
sudo su - dsmr
./deploy.sh
```
Result:
<details>
  <summary>Traceback (most recent call last):</summary>
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above

The above exception was the direct cause of the following exception:  

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 89, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/commands/migrate.py", line 75, in handle
    self.check(databases=[database])
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 419, in check
    all_issues = checks.run_checks(
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/checks/registry.py", line 76, in run_checks
    new_errors = check(app_configs=app_configs, databases=databases)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/checks/model_checks.py", line 34, in check_all_models
    errors.extend(model.check(**kwargs))
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/models/base.py", line 1303, in check
    *cls._check_indexes(databases),
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/models/base.py", line 1695, in _check_indexes
    connection.features.supports_covering_indexes or
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/postgresql/features.py", line 93, in is_postgresql_11
    return self.connection.pg_version >= 110000
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/postgresql/base.py", line 329, in pg_version
    with self.temporary_connection():
  File "/usr/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 603, in temporary_connection
    with self.cursor() as cursor:
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 218, in ensure_connection
    with self.wrap_database_errors:
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above

</details>
`psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above`

### Workaround
Voor nu is dit de workaround `scram-sha-256` aanpassen naar `md5`:
/etc/postgresql/14/main/postgresql.conf
'password_encryption = md5'
&
/etc/postgresql/14/main/pg_hba.conf
'METHOD' (gehele kolom) aanpassen naar 'md5'

sudo -u postgres psql -c "alter user dsmrreader with password '<password>';"
service postgresql restart

Refrence:
https://dothanhlong.org/fix-unable-to-connect-to-postgresql-server-scram-authentication-requires-libpq-version-10-or-above/

Zou je deze op de to-do lijst willen zetten om te fixen?

### DSMR-reader version

5.1

dennissiemensma, 2022-05-08:
Bedankt voor je melding. Ik zal hier in de toekomst een keer dieper naar kijken. Zelf heb ik postgresql 14 nog niet geprobeerd.

dennissiemensma, 2022-06-08:
Met Docker lijkt het iig geen issue te zijn. Lokaal even Postgres 14 gedraaid met de laatste DSMR-reader. Daar staat het op `host all all all scram-sha-256`.

Als ik wat verder zoek dan kom ik hier: https://stackoverflow.com/a/62808487
Heb je ook al geprobeerd om `libpq-dev` te installeren op Ubuntu en vervolgens (pas) DSMR-reader inclusief virtualenv? En anders kun je het voor nu laten voor wat het is.

Op termijn kun overstappen naar de Docker-variant. Uiteindelijk zal ik de handmatige installatie niet meer actief ondersteunen, juist omdat er zoveel variaties en versies zijn.

dennissiemensma, 2022-06-08:
Ik laat dit voor nu even voor wat het is.

closed 2022-06-08 by dennissiemensma