dsmrreader/dsmr-reader #1628: 💡MQTT bericht type

onsmam, 2022-05-09:
### Feature

HI,

is het mogelijk om het bericht type in te stellen op een mqtt topic?
wellicht stel ik mijn vraag niet duidelijk genoeg dus zal hem even toelichten.

ik wil de data gaan gebruiken om eea richting een solar edge omvormer te gaan pushen. ([Meter proxy](https://github.com/nmakel/solaredge_meterproxy)
blijkbaar stuurt DSMR reader alle waardes als string, echter heeft de meter proxy de data als floats nodig.
is het mogelijk in te stellen? of evt te veranderen?

alvast bedankt!

Jacques

dennissiemensma, 2022-05-09:
Bedankt voor je melding. Dit is toevallig al gevraagd in https://github.com/dsmrreader/dsmr-reader/issues/1531#issuecomment-988232586. 

Het is helaas niet makkelijk om te veranderen, omdat dat alle andere integraties raakt die er nu al mee werken. Het is doorgaans wel vrij triviaal om dat aan de ontvangende zijde te doen, zonder dat het dingen stuk maakt. Echter is daar geen wereldwijde consensus of standaard voor, ivm floating point, dus geen garantie dat dat wel in bijvoorbeeld Meter Proxy gedaan gaat worden.

closed 2022-05-09 by dennissiemensma

onsmam, 2022-05-09:
duidelijk, dankje wel!

het is een python script, volgens mij als ik overal float() voor zet moet het goedkomen, ga dat proberen!