dsmrreader/dsmr-reader #1630: Data uit Domoticz geimporteerd. Geen grafieken voor electriciteit.

Flappiee, 2022-05-11:
### Description

Ik heb mijn Domoticz data geimporteerd in DSMR Reader.
Hiervoor heb ik dit powershell script gebruikt: https://github.com/dsmrreader/dsmr-reader/issues/520

Het import proces was redelijk snel klaar.

Als ik nu in mijn " Archief"  kijk zie ik wel historisch gas verbruik, maar geen elektra verbruik.
Als ik onder de configuratie naar Verbruik en dan Electriciteitsverbuik ga zie ik hier per dag wel mijn data staan, maar alles staat op 0.
Als ik een datum hierin aanklik zie ik wel de totalen van die dag.

Is dit normaal en moet dit nog "verwerkt"  worden? of gaat er iets fout?

### DSMR-reader version

5.1

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.1 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                               3886
    Consumption records electricity / gas (est.)                                                  3815 / 3815
```

Flappiee, 2022-05-11:
Zie nu waar met misgaat.
Het PowerShell script maakt per dag een meting aan op 00:00 en op 23:59, tenminste, dat is de bedoeling.
Als ik in DSMR Reader kijk zie ik metingen om 00:00 en 00:01. Dus voor elke dag 1 minuut....

[image]

Iemand dit eerder gezien met dit script?

dennissiemensma, 2022-05-11:
Ik kan je hier helaas niet mee helpen. Je kunt even zoeken in oude issues over Domoticz:
- https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+Domoticz+is%3Aclosed+sort%3Aupdated-desc

Zoals:
- https://github.com/dsmrreader/dsmr-reader/issues/908#issuecomment-592456012
- https://github.com/dsmrreader/dsmr-reader/issues/908#issuecomment-1003215754

Flappiee, 2022-05-12:
Dank je wel voor je reactie.
In #908 staat inderdaad helemaal onderin de oplossing:

_Ik zag alleen het Gas maar niet de Electra.
Toen ik de database tabel dsmr_consumption_electricityconsumption ging inspecteren kwam ik er achter dat alle timestamps er 1 minuut erbij hadden gekregen. (in dsmr_consumption_gasconsumption was dat niet)_

Wel vreemd, want het script geeft wel de juiste timestamps mee, maar DSMR reader lijkt ze verkeerd te verwerken?

Ik heb dus dit gedaan zoals aangegeven in #908 :

sudo supervisorctl stop dsmr_backend
sudo supervisorctl stop dsmr_datalogger

sudo su - postgres
psql dsmrreader
update dsmr_consumption_electricityconsumption set read_at = read_at - interval '1 minute';
(Response: UPDATE 3894)
truncate dsmr_stats_daystatistics;
(Response: TRUNCATE TABLE)

exit

sudo supervisorctl start all

Nu loopt het proces voor de dagstatistieken dus opnieuw. Even afwachten wat er nu uitkomt :)

maar waarom worden de gegevens verkeerd geimporteerd in DSMR Reader? Is dta misscien een bug?

dennissiemensma, 2022-05-13:
Is het inmiddels gelukt? Het is ongetwijfeld een bug, maar het is lastig te zeggen waar het in zit. Het kan in het script zitten, in de tijdzones, in het achtergrondproces dat wel of niet uit moet tijdens de import, in de API, etc etc.

Ik zou persoonlijk gewoon de pragmatische fix toepassen, helemaal omdat het eenmalig is.

dennissiemensma, 2022-05-17:
@Flappiee Is het inmiddels gelukt?

Flappiee, 2022-05-18:
@dennissiemensma, ja is dus gelukt. Na de import staat alles dus verkeerd en kun je op deze manier de tijden goed zetten.
Daarna opnieuw de dag statistieken laten berekenen:

sudo supervisorctl stop dsmr_backend
sudo supervisorctl stop dsmr_datalogger

sudo su - postgres
psql dsmrreader
update dsmr_consumption_electricityconsumption set read_at = read_at - interval '1 minute';
(Response: UPDATE 3894)
truncate dsmr_stats_daystatistics;
(Response: TRUNCATE TABLE)

exit

sudo supervisorctl start all

dennissiemensma, 2022-07-05:
# Update
Missende dagen in Archief?

Probeer:
```
sudo su - dsmr
./manage.py dsmr_stats_reconstruct_missing_day_statistics
```

#1654